looooo/freecad.gears #1: Worm Gears

hamish2014, 2015-07-11:
Hi,

Firstly I want to thank you for the gear workbench which I have been finding very useful.

My issue has to do with trying to create a worm gear.
For which I get funny edges
[image]

Allouth not catastrophic, these edges make 3D printing the worm gear rather messy.

looooo, 2019-09-24:
Here is a better image:
[image]

looooo, 2019-09-24:
I guess a involute gear shouldn't be the base tooth geometry for a worm gear. I guess the profile is more like a rack:  
[image]

looooo, 2019-09-24:
for sure, but someone has to do the work. This is an example of the profile which we need for the helical extrusion. It's basically a helical projection on the side face of the rack. As you can see it looks symmetric and so only two parts of a circle and one bspline are necessary to create the basic profile:
[image]

I used a rack to create the worm gear and afterwards split the rack with a box to obtain the profile. Here is the file:
[worm-gear.zip](https://github.com/looooo/freecad.gears/files/3647222/worm-gear.zip)

simple replace the ending of the file to "FCStd" to open it in FreeCAD.

Foadsf, 2019-09-25:
> but someone has to do the work

If you had Patreon, Flattr or Liberapay, I would support you and advertise as I have done for others [here](https://www.reddit.com/r/opensource/comments/ckkl5e/supporting_free_libre_and_open_source_cad/?utm_source=share&utm_medium=web2x).

looooo, 2019-09-25:
Feel free to test:
https://github.com/looooo/freecad.gears/commit/edc1d9a1711b89ad3abb5aa4b49feaa395b6e449
[image]

looooo, 2019-09-28:
> If you had Patreon, Flattr or Liberapay, I would support you and advertise as I have done for others here.

Feel free to support me ; )
https://github.com/looooo/freecad.gears#a-gear-module-for-freecad

Foadsf, 2019-09-29:
@looooo added you to [the list](https://www.reddit.com/r/opensource/comments/ckkl5e/supporting_free_libre_and_open_source_cad/?utm_source=share&utm_medium=web2x) and will advertise for you on Twitter. Though IMHO it would be nice to share your actual name and identity so people who are they contributing to. 
I also updated the workbench through FC Addon manager to test the new `worm_gear`, but nothing was added to the icons as far as I can see.

looooo, 2019-09-30:
Thanks for the feedback. I have added some personal content.
> I also updated the workbench through FC Addon manager to test the new worm_gear, but nothing was added to the icons as far as I can see.

I am working on a develop branch. So if you use the addonmanager with git, simple change to the .FreeCAD/Mod/FCGear directory and switch the branch:
`git fetch --all && git checkout develop`

Foadsf, 2019-09-30:
@looooo This seems like a nice feature to ask. inside the addon manager, one can select the branch, release and even commits. 🧐

sgrogan, 2019-10-01:
I think something like this, from FreeCAD python console, should work?
```
import git
repo = git.Repo(FreeCAD.getUserAppDataDir()+'/Mod/FCGear')
repo.git.pull('origin','develop')
repo.git.checkout('develop')
```
unfortunately, a FreeCAD restart is required.

Symbian9, 2019-10-03:
FTR, latest [*SolveSpace*](https://github.com/solvespace/solvespace) nightly builds got «Helix» feature, but it still not perfect and there is discussion on its forum:

- http://solvespace.com/forum.pl?action=viewthread&parent=2831

On this discussion linked «Thread for Screw Tutorial» on *FreeCAD* wiki:

- https://www.freecadweb.org/wiki/Thread_for_Screw_Tutorial

looooo, 2019-10-06:
btw.: The worm gear is based on the idea of @mwganson: https://github.com/mwganson/ThreadProfile But instead of using only one bspline multiple curves are used.

closed 2020-09-22 by looooo