looooo/freecad.gears #6: Internal/ring gears?

pperrin, 2017-01-31:
Not sure this is the place to put questions, but...

Is it possible to generate internal ring gears for planetary gear systems?

I can generated them with other software, and if the maths is right they should work(!) - but it would be neat to do it all in the place.

looooo, 2017-02-01:
it's possible, to do them with part-wb. (Make a ring and subtract an inner gear) The problem is that the gears have no parameter for the height of the head....
[image]
But you can also use functionality of part-design: change to part-design-wb:In top menu choose Part-Design and select the Involute Gear at the bottom. There you have the option to choose an inner or outer profile.

pperrin, 2017-02-01:
Thanks for the suggestions - I understand the inner ring teeth are a different profile so the 'subtraction' method doesn't work well.

The built in gears come out just too big when 3d printed and don't have any parameters to adjust to give them a bit more tollerance... but I'll try the built in ring gears and see if your workbench has enough parameters to do all the adjusting on the star/planets...

Thanks

looooo, 2017-02-03:
I have added 2 new properties to the involute gear. With these you can manipulate the tooth-hight and also make the tooth thicker (reversed backlash). With this it's possible to create inner gears...

[image]
[image]

pperrin, 2017-02-03:
Cool - I have installed the new files - I don't know when I'll be back on gears, but when I am I'll let you know how it goes.

Great Job - thanks.

alexisdal, 2017-04-17:
hello,
First of all, thanks for this excellent piece of software directly integrated in freecad. very convenient indeed. 

i'm trying to use it to design some planetary gears as well but i'm also facing the same issue with the ring/annulus.

It seems the new parameters (reverse backlash and head) should do the trick.

However, i am somehow suprised by the value of bashslash. for a 12-36-96 planetary gears (in teeth count for sun, planet and ring espectively). the -- apparently correct -- value i found are
0.1mm of head 
0.006m reverse bashlash

head value fees ok.
however, bashlslash feels very very tiny. is it normal ?

looooo, 2017-04-18:
>0.006m reverse bashlash
Maybe you  meant 0.006 mm?
I think the head-parameter is multiplied with the module. As I thought backlash is more related to machine-tolerance and not dependent on the module (diameter) it's given by an absolute-value. If there is any good documentation (or standardization) regarding these parameters I can update the geometry-influence of these parameters.

kcleung, 2017-07-01:
A spline on a CV shaft is a good example of a tooth axle:

https://en.wikipedia.org/wiki/Spline_(mechanical)

kcleung, 2017-07-01:
And when we make an internal gear, the gear may well also have an external gear too.

sdfgeoff, 2021-01-04:
It is worth noting that the solution above (adjusting backlash etc.) works for when the interior/exterior gears differ only slightly in tooth count, it does not work well when the tooth counts differ significantly.

In the end I decided to go with cylindrical gears, so I could use the built in involute gear tool.

looooo, 2021-01-04:
> It is worth noting that the solution above (adjusting backlash etc.) works for when the interior/exterior gears differ only slightly in tooth count, it does not work well when the tooth counts [...]

Can you explain? maybe add some Images to visualize the issue.

sdfgeoff, 2021-01-04:
Disclaimer: I'm no expert on gears. If someone more knowledgable says otherwise, go with them.

Here is a 24 ring gear and a six tooth pinion  (yeah, I know you shouldn't make them that tiny) I've already got a bit of negative backlash in here, but would need a fair bit more to give adequate clearance.
[image]

Here's a 9 tooth pinion to a 24 tooth ring gear:
[image]

There's no backlash/head in the model in this demo, and you can see that I'd need to adjust the amount of compensation to be about half the width of the "tooth" on the ring gear (green) which would result on the loading being on the tooth "flank" (yellow  arrow)
If the tooth profile was flipped (cyan) then the loading would be on the tooth flank (I think).

Possible useful reference (I'll have a more detailed look at this tomorrow, but at the very least it seems to hint that there is a difference): https://gearsolutions.com/departments/tooth-tips/internal-ring-gears-design-and-considerations/

looooo, 2021-01-05:
Does a bottom-fillet-radius property solve this issue? This is on the list for a long time.

looooo, 2021-01-05:
Btw. Try to set the clearance property of the inner gear to a negative number to cut the intersecting part. Or use a Tube-Feature of the Part-Workbench to increase the inner radius.

[image]

sdfgeoff, 2021-01-08:
Hmm, I didn't try the clearance option. That does indeed seem like it would fix it. An bottom fillet option may be nice, but with the clearance parameter, I'm not sure if it's needed.

jrmoserbaltimore, 2021-02-13:
Is it difficult to implement a property that just turns an involute or cycloid gear into a ring gear?  (In the scale that implementing anything like this is difficult)

looooo, 2021-02-13:
@jrmoserbaltimore if there are special properties necessary for the internal-gear I think it's best to create a new feature. For now please use a workaround with boolean operation eg. Part-workbench

jrmoserbaltimore, 2021-02-13:
Thanks.  I wasn't sure if this had been decided the appropriate approach or if it's just a for-the-future thing.

I guess generating an entire planetary gear set at once would make some sense, but be hard, and take a while to implement.

looooo, 2021-06-05:
Btw. with the addition of fillets I don't see any issues with creating internal gear rings.

closed 2021-06-05 by looooo