looooo/freecad.gears #7: Screenshots and more elaborate description

luzpaz, 2017-05-24:
Hey @looooo thanks for your work on this addon! I'm volunteering to help improve the addon manager.  
Would you mind:  
- [ ] adding some screenshots to the README
- [ ] tagging this repo with the 'freecad'   
- [ ] offering a FreeCAD forum link for folks to ask/discuss issues related to this addon

looooo, 2017-05-24:
I currently do not have much time for this module. But if I find the time I will try to work on this. Pr are also appriciated...

luzpaz, 2019-05-20:
Opened https://github.com/looooo/freecad.gears/pull/25  
There are still remaining issues. Perhaps anyone else (besides looooo) would like to help?

greenpdx, 2020-12-27:
Good package! I come from the electrical side of engineering. 
Lanterngear: 
I am trying to create a gear for a 1/4 (6.35mm) pitch roller chain with a  roller diameter of 0.13(3.3mm).
module = pitch/PI  6.35/Pi = 2
bolt_radius of 1.65  = 3.3/2

Yes this correct?  
Please document this formula

luzpaz, 2020-12-27:
@greenpdx JFYI, your request is off-topic of this ticket.

greenpdx, 2020-12-27:
Thought this was for better documentation.  This was an example of what is needed.

luzpaz, 2021-08-03:
@looooo do you mind also adding a reference to a tutorial or 2 that you endorse for this WB ? Maybe also tutorials that show how to use in an one or more of the various assembly workbenches?

closed 2024-10-31 by luzpaz