looooo/freecad.gears #8: helical gear angle option for involute and bevel gears

kcleung, 2017-06-24:
Thank you very much for setting up the FCGear module.  At the moment, it has the InvoluteGear,  rack, cycloide hear and bevel gear.

At the moment, teeth are limited to spur option, yet a lot of gears we see are single and double helical.

It would be great if we can add these two parameters to allow creation of helical gears / racks:

number of helix (1 or 2)  - if 2, then double helix with opposite angles
angle of tooth (if between 0 to 90, then it is helical)

looooo, 2017-06-24:
helical are done with the beta-property. I can add a beta-property to the rack.

kcleung, 2017-06-24:
Thanks looooo!  I just tried the beta-property, and I successfully set up a single helix gear for involute and cycloide gears.

However, I can't see the beta-property in bevel gears, could you please add it to bevel gears too?

Also, for involute, cycloide and rack gear, could you please add the double helix option too?  Double helix is so common in gears and racks.

looooo, 2017-06-24:
double helix is made by mirroring the gear with part. (pretty simple) I don't think having a property for this is necessary.
Helical bevel-gears, are difficult. I never succeeded to create them. Also i do not have any documents showing how the maths behind the geometry... But I know helical-sphericalinvolute gears exist.

kcleung, 2017-06-24:
But having a property for double helix will certainly speed up the process and simplifies the user's procedure a lot.  All we need is to add an extra boolean parameter into the gear class.

After all, the purpose of the gear workbench is to automate creation of gears, otherwise, one could say that the gear workbench as a whole is unnecessary as user can draw gear using the part workbench and use the polar feature to copy all the teeth!!!!!

In the future, we will want to have the gear train assembly feature, so it is good to get double helix option as a parameter, so that the solver can automatically create the desired gear train in a click!!!!!!!

Therefore I see double helix as an *essential* feature in the gear workbench, and all we need is to add a checkbox in the gear parameter.

kcleung, 2017-06-25:
I am implementing the double helix function for involute gear.  I am having a good look at helicalextrusion() function in gearfunc::_Classes.py

def helicalextrusion(wire, height, angle):
    spine = Wire(Line(fcvec([0., 0, 0]), fcvec([0, 0, height])).toShape())
    auxspine = makeHelix(height * 2 * pi / abs(angle), height, 10., 0, bool(angle < 0))
    solid = auxspine.makePipeShell([wire], True, True)
    return solid

Here I notice that the local variable "spine" is not used anywhere in the code.  Are there any reasons why the spine variable is defined?

In my implementation, I plan to make helicalextrusion take an extra boolean parameter "double_helix".  If double_helix is true, then this function will return a solid with a double helical gear.

In this case. do I make two solids, first solid has beta angle, 0.5 * height, and secnd solid has - beta angle, -0.5 * height, and then union them together into a single solid with the Part workbench code and return the unified solid?

If so I will try, and the put it into my branch for you to have a look!

kcleung, 2017-06-25:
This is my attempt for a double helix, by halving the height of the first solid, and then mirroring it:

def helicalextrusion(wire, height, angle, double_helix = False):
    if double_helix:
        direction = bool(angle < 0)
        first_spine = makeHelix(height * 2 * pi / abs(angle), 0.5 * height, 10., 0, direction)
        first_solid = first_spine.makePipeShell([wire], True, True)
        second_solid = first_solid.mirror(Base.Vector(0,0,0.5 * height), Base.Vector(0,0,1))
        return first_solid.fuse(second_solid)
    else:
        spine = Wire(Line(fcvec([0., 0, 0]), fcvec([0, 0, height])).toShape())
        auxspine = makeHelix(height * 2 * pi / abs(angle), height, 10., 0, bool(angle < 0))
        solid = auxspine.makePipeShell([wire], True, True)
        return solid

The shape looks correct.  However, fuse() takes up to 5 minutes, and do you know why the resultant gear looks black?

looooo, 2017-06-25:
Have you tried using the part workbench mirror. Maybe looking at the implementation of this function gives some hints how to implement it.
For sure the fusion operation is quite time-intensive. If there is no nead to fuse the objects (which is true in the most use-cases) I think also a compound could be an option. As there is no joining of faces involved, the compound command is much faster, and should never result in a black solid.

>The shape looks correct. However, when double_helix is set to True, fuse() takes up to 40 seconds with 100% CPU resource, and do you know why the resultant gear looks black?

A black solid is a inverted solid. (inner normals pointing outwards because of fusion problems...) I can't tell why this happen for your code. For me the fuse command from part-workbench creates a valid solid...

> Here I notice that the local variable "spine" is not used anywhere in the code. Are there any reasons why the spine variable is defined?

This is from an older implementation. This line should be removed.

kcleung, 2017-06-25:
from the code I wrote above, am I using the part workbench mirror?

first_solid = first_spine.makePipeShell([wire], True, True)
second_solid = first_solid.mirror(fcvec([0,0,0.5 * height]), fcvec([0,0,1]))

Here, makePipeShell returns a TopoShape object, and is its mirror method a Part workbench mirror function?

Also, is the TopoShape::fuse() that I used a Part workbench fuse method?  I am confused

kcleung, 2017-06-25:
Just tried to do the mirror and fuse at the Part workbench GUI.  The fuse takes much less time, and does not result in a black object.  However, how can I have access to the Part mirror and fuse API calls?

kcleung, 2017-06-25:
I just revised my code:

def helicalextrusion(wire, height, angle, double_helix = False):
    if double_helix:
        direction = bool(angle < 0)
        first_spine = makeHelix(height * 2 * pi / abs(angle), height, 10., 0, direction)
        first_solid = first_spine.makePipeShell([wire], True, True)
        second_solid = first_solid.mirror(fcvec([0,0,0]), fcvec([0,0,1]))
        return first_solid.fuse(second_solid)
    else:
        spine = Wire(Line(fcvec([0., 0, 0]), fcvec([0, 0, height])).toShape())
        auxspine = makeHelix(height * 2 * pi / abs(angle), height, 10., 0, bool(angle < 0))
        solid = auxspine.makePipeShell([wire], True, True)
        return solid

This code now creates a double helix much faster, and no longer result in a black solid.

So I guess it was my fault and I somehow messed up with the dimensions.

However, the height of the double helix gear becomes double the original gear.  Also, the mirroring only mirror from the XY plane at z=0 regardless of the actual position and orientation of the gear.

Should I still commit it in my fork, and then deal with these issues in later commits?

looooo, 2017-06-26:
very good. thanks for the contribution.
I added your commits, and changed the fusion to a compound. This way the computation is much faster.

>However, the mirroring only mirror from the XY plane at z=0 regardless of the actual position and orientation of the gear. Could you please have a look at my code and suggest how can the mirroring step take account to the actual position and orientation of the gear?

This is not necessary. The geometry is always the same (extrusion in z-direction) The transformation is applied afterwards. So there is no problem with setting some transformation on a double-helical-gear...

kcleung, 2017-06-26:
> I added your commits, and changed the fusion to a compound. This way the computation is much faster

However, if it is a compound, then will the double helix gear still be able to be exported as one solid for IGES, one mesh for STL, and one body for our future assembly support?

If not, then we better use something else that satisfy these requirements.

looooo, 2017-06-26:
>will the double helix gear still be able to be exported as one solid for IGES, one mesh for STL, and one body for our future assembly support?
I don't know, but if it is, there is most-likely a bug in another part of freecad. 
To me it doesn't make much sense to use fuse as the output isn't different from a compound. Fuse is more for overlapping parts.

kcleung, 2017-06-26:
so the question is whether a compound is an object, or a group of objects?

looooo, 2017-06-26:
In the end I think it doesn't make much sense to add a double helix parameter. It's really simple to create the double helix-gear with the gui.
Better to work towards a general gear adapter-feature which can add gear parameter to any object. (To be later used by assembly)

kcleung, 2017-06-26:
I checked, it is a single solid, and also you are right that it works regardless of what angle the gear is at.  When I said double helix, I mean the overall height is the same.  I found a bug in the helicalextrusion() where the single helix has height accidentally halved.  This is fixed.  I also attempted to fix the problem where the double helix is shifted to -0.5 * height because of extruding half height and then mirroring.  The fix was by calling translate() on the compound, but it hasn't appeared to work.
It is in my fork, could you please have a look?

When you say the "gear adapter feature", you mean that a row of teeth can be added to any surface, including the exterior and interior of a cylinder, and rectangular surface?   This sounds interesting, though we need to spend good time on its design details.

looooo, 2017-06-26:
> When you say the "gear adapter feature", you mean that a row of teeth can be added to any surface, including the exterior and interior of a cylinder, and rectangular surface? This sounds interesting, though we need to spend good time on its design details.

No this should only be an adapter to add constraints between two gears.

to solve:
> In the future, we will want to have the gear train assembly feature, so it is good to get double helix option as a parameter, so that the solver can automatically create the desired gear train in a click!!!!!!!

kcleung, 2017-06-26:
Get you!  so I will need to have a good look at constraints data structure and APIs in FreeCAD.

Let's get back to the topic:

for all gears, is the position the centre of the gear (x, y and z) on all three axis or at the bottom end of one or more axis?  This has to be consistent for all gears, involute or helix.

I have just fixed the height and translation problems in pull request #13.  single and double helix now work in any angle and positions, and neither position or height are affected by switching from single to double helix and vice versa.

looooo, 2017-06-27:
Thanks for the PR.

>I have just fixed the height and translation problems in pull request #13. single and double helix now work in any angle and positions, and neither position or height are affected by switching from single to double helix and vice versa.

I should have looked more carefully at the difference between the helical-extrusion function. I think I have reverted some changes previously, sorry...

Now the rack also needs a beta and double helix property. Helical gears with same module, are made from the same racks but with the rack turned about the value of beta. This means the 2d-profile of the rack has to be scaled in x with the cos of beta. The best way to achieve the extrusion is maybe a loft. For the double helix we could use one loft with 3 wires, I think this should be possible.

looooo, 2017-06-27:
I have added the double helix rack added:
[image]

looooo, 2017-06-27:
maybe we can use a sweep with more then one profile for the helical extrusion of the involute gear. This way the computation should be fast and the result is one solid. Similar to the double helix rack where a loft with 3 wires is used.

kcleung, 2017-06-27:
That's good work!  I have tried to pull your changes to my fork.  However, it says there are merge conflicts.  Then when I had a look at the commits, it seems that when you execute my pull request, the commits in my pull requests are somehow altered when they appear in your repository despite the diffs are identical.  For example:

The "cleanup" commit in my pull request is 941f8ff.  However after you execute my pull request, the SHA digest of this commit becomes 38d03df in your repository.  This makes it very difficult for me to merge your newest changes to my fork.  However, this does not happen in any other repositories that execute my pull request.  Do you know the cause of this problem? 

We better sort this out, or it will be very difficult for any other developers working at this FCGear plugin

kcleung, 2017-06-27:
I just wonder how you executed my pull request?  Did github automatically merge my branch into your branch, or did you manually pull my fork into your local branch (at your local computer), and push to your branch by fast forward?

looooo, 2017-06-27:
I used the rebase option of github. Maybe I have pushed with fast-forward for the last commit. But I have no idea why it should have influence on your commit.

kcleung, 2017-06-27:
I thought fast-forward should not change the SHA digest.  It just pushes the thing you pulled from remote verbatim. And if SHA digest is changed, it confuses git, and make merging so difficult.

For example, I make a fork from the offical FreeCAD repository https://github.com/FreeCAD/FreeCAD

https://github.com/kcleung/FreeCAD

I have a local copy of FreeCAD in my computer with this .git/config:

[core]
	repositoryformatversion = 0
	filemode = true
	bare = false
	logallrefupdates = true
	ignorecase = true
	precomposeunicode = true
[remote "upstream"]
	url = https://github.com/FreeCAD/FreeCAD.git
	fetch = +refs/heads/*:refs/remotes/origin/*
[remote "origin"]
	url = https://github.com/kcleung/FreeCAD.git
	fetch = +refs/heads/*:refs/remotes/origin/*
[branch "master"]
	remote = origin
	merge = refs/heads/master

Then whenever I want to merge the official FreeCAD repository into my fork, then this is what I do at the command prompt of my mac/linux machines:

$ git pull remote master
$ git push origin master

As you can see, the current master head commit of the official repository is  86c8004:

https://github.com/FreeCAD/FreeCAD/commit/86c8004802f0c69555f49468b6008747f6e551b1

and in my fast-forward push into my fork using the above two git commands, the SHA1 sum stays at:

https://github.com/kcleung/FreeCAD/commit/86c8004802f0c69555f49468b6008747f6e551b1

and my fast forward action did not change the SHA hash.

One more thing I notice:  In the above fast-forward, the commit shown in my fork stays as "yorikvanhavre Arch: Small fix in Rebar"

However, in the FCGear case, in my fork, commits in my fork says " kcleung committed a day ago ", but after you pull into your branch, it becomes " kcleung committed with looooo a day ago "

Could your process somehow inserted "with looooo" and therefore changed the SHA digest?

We better sort this out now, as it will be totally unacceptable to ask developer to reset their fork every time after a pull request is executed.

looooo, 2017-06-28:
>However, in the FCGear case, in my fork, commits in my fork says " kcleung committed a day ago ", but after you pull into your branch, it becomes " kcleung committed with looooo a day ago "

yorik is owner of the FreeCAD-repo... So this was a bad example. Try to have a look at commits of someone else. You will see its' the same as for your commits to my repo.

I am pretty sure the merge-conflict is not caused from the way how the PR was merged, but from a 
`git commit -a --ammend` which I have `git pushe origin +master`. Your commits are regularly merged. Compare them with the commits of other users of the FreeCAD-repo for example. (the numbers simple doesn't match, and the person who accepted a PR is always listed... I think you are making wrong assumptions.

Have you tried to see at which commit the merge conflict has happened?

kcleung, 2017-06-28:
when I run:

git pull upstream master, it complains there is merge conflict on _Classes.py that cannot be automatically resolved, as it fails to see my last three commits as the same as your first three commits despite they are identical.  

The correct behaviour should be to acknowledge the three commits "cleanup", "fixed accidental translation... " and fixed height bug...." in both forks as the same thing (they have different digests), and fastforward to your tip, but git can't do that.  Here is the screenshot on my local copy, where pink represents my fork, and blue represents yours:

[image]

To replicate my problem, you can do the following at the command prompt on your computer:

$ git clone https://github.com/UoA-CompSci/FCGear.git
$ cd FCGear
$ git pull https://github.com/looooo/FCGear.git master

Since I do not have any further commits after I sent you the pull request, the correct behaviour should be to fast-forward the local repository to your repository with the new changes you committed after executing the pull request, but instead, the system fails to see the three commits mentioned above from my branch (your local copy) as the same as commits pulled by you, as the digest is altered.  Therefore it sees the forks diverges from the point I make the fork, and forces me to manually resolve the merge by teasing out which bits are old and which bits are new.  This is unacceptable as this manual process will be very prone to human errors, while the merge should just be a simple fast forward.

looooo, 2017-06-28:
I am no git expert, but I always use the rebase command to get up to date with another branch. This adds my additional commands at the top. In your case there are no such additional commands, but rebase simple does the job, while the pull fails.

To avoid such situations in the future, I will try to work with pull-requests.

For now I think you should simple do a `git reset remote master --hard` or however you have named this remote repo.

kcleung, 2017-06-28:
To get up to date with another branch, the correct command should be:

$ git pull <repo> <branch>

If there are conflicts that cannot be automatically merged, then do a manual merge and commit at this point (usually there isn't).

and then push it to the master branch of your github repository:

$ git push origin master

Rebase is for rebasing *your* changes to the tip of the tree of the master repository, so it is the wrong tool to use where your repository is the master.  More importantly, rebase should only be used where the changes are only in your computer and *not* published yet.  Once any commits are published, rebase must not be used, and you should only use "merge".

When you run "rebase" is this what you encounter

                   G - H - I                        my fork
                / 
A - B - C - D - E - F                        your repository

where by the time you look at my pull request I, you have already committed D E and F to your repository?

Therefore, if G, H, I are already published, git reset must *not* be used, or it will wreak havoc on the published repository!  Instead, we should use merge:

                   G - H - I                            my fork
                /                \ 
A - B - C - D - E - F - K                       your repository

where "K" is the merged commit.

I tried your command:

$ git reset remote master --hard 

where I named your repository as "remote" but it fails and is not going to work, because the three commits in the pull request (that has different SHA from your amended version) is already on my github fork, and I have no way to get rid of the these three commits on my fork, short of deleting my repository and start again.

So now I have no choice but to delete my repository and re-fork your one.

From now on, we must not use the "rebase" command on published repository.

looooo, 2017-06-28:
> $ git reset remote master --hard
can you show the output? This really shouldn't fail.

kcleung, 2017-06-28:
see my edited comments above.  even if it does not fail, there are no ways for me to get rid of the three commits in the PR that has different SHA and already published

looooo, 2017-06-28:
ok I see the problem. But it works for me if I first reset on your master:
`git reset origin master --hard`
and then reset on my branch:
`git reset remote master --hard`

kcleung, 2017-06-28:
Still doesn't work for my local working copy:

 FCGear abc$ git --version 
 git version 2.11.0 (Apple Git-81)
 FCGear abc$ git reset origin master --hard
 fatal: Cannot do hard reset with paths.

So which version of git you use and what is your OS?

Also, even if it works on my local working copy, there are still not way to get rid of the three commits on my fork which is already published at github, so the only way would be to delete my fork and re-fork from your one.

In the future, we must not use rebase on any published commits.

looooo, 2017-06-28:
Sry, my fault. It should be:
`git reset remote/master --hard`

>Also, even if it works on my local working copy, there are still not way to get rid of the three commits on my fork which is already published at github

For sure there is:
`git push origin +master`

kcleung, 2017-06-28:
This is what I did:

$ git reset origin/master --hard
$ git push origin +master

So it now converts commits in my original pull request into your amended one, and it happily accept your new commits after push request.  Please have a look at my fork now:

https://github.com/UoA-CompSci/FCGear/commits/master

Does this look good to you?

Now I can try out your new work, thanks!

looooo, 2017-06-28:
looks good. I have changed some things locally again. There is a problem with the mirroring of the helical extrusion. Somehow the two solids are kind of overlapping. You can see this if you zoom in...
So I changed the method to use the z=0 plane to mirror and afterwards translate the combined solid.

So I will commit to another branch, push this branch to github, create a PR and merge it. Please give feedback if this is working for you.

looooo, 2017-06-28:
With using the merge option the sha-number stays the same. The only thing that is not that nice is the additional merge-commit https://github.com/looooo/FCGear/commit/e419f55709435575e05e43cca866b2371e82ab9c
This is the different for the rebase-option. There the sha-number is changed, but the commit is only listed once.
regarding merge vs rebase: https://www.atlassian.com/git/articles/git-team-workflows-merge-or-rebase

kcleung, 2017-06-28:
It works for me (and other contributors too)!  

As what the article says, if the new commits of the develop branch and the concurrent master commits are not yet published, it does not matter whether you rebase or merge, but once these commits are published, then you are pretty much stuck with merge.

looooo, 2017-06-28:
published or locally what's the difference?

kcleung, 2017-06-28:
published = commits pushed into github or other internet-viewable sites such as bitbucket (your repository or any other repositories)

locally = commits that stays in the working copy on your computer only  (that you haven't pushed to any other places yet)

looooo, 2017-06-29:
For sure, but what is the difference of published or local regarding the use of merge vs rebase?

>When you run "rebase" is this what you encounter...
where by the time you look at my pull request I, you have already committed D E and F to your repository?

This was not true for our situation. I merged your PR pulled it and added my commits. The only problem was the forced push because I changed one published commit. I think you are making wrong assumptions!

>Rebase is for rebasing your changes to the tip of the tree of the master repository, so it is the wrong tool to use where your repository is the master

the rebase option of github rebases the PR-branch on master (which simple adds the additional commits to the tip of master, same as doing `git rebase remote master` locally) and reset master to the PR-branch. I think you got that wrong.

In the end git is a tool to help with collaborations. I made a mistake, and you got a merge-conflict. Such situations can occur and git has the tools to solve them...

looooo, 2017-06-29:
This commit https://github.com/looooo/FCGear/commit/962c0e4ffeaf8b97c43ec73d3b0055945d3940d9 was merged with the rebase option. (As you can see the commit history only contain one commit, no merge commit). If I pull this commit back into my develop branch (locally) git wants to add a empty  merge-commit locally.

kcleung, 2017-06-29:
I have pulled this into my fork without any problems:

git pull upstream master
git pull origin master

so I guess as long as the commits in your master branch and rebase branch are all yours, then you are also fine.

looooo, 2021-06-05:
closing this one, as double helix is available for the main-gear types

closed 2021-06-05 by looooo