looooo/freecad.gears #9: internal gear

kcleung, 2017-06-24:
It would be great to be able to create internal gear (involute or cycloid teeth), as this would allow the creation of a planetary gear train (by a different function).

I guess it would be great for a piece of gear, it can have two set of gear data, namely internal and external.  The one which is not used can be set to null.

Apart from allowing the possibility of creating a planetary gear assembly, the internal gear function can also allow a gear to slide on a toothed axle:

toothed axle - involute external gear with very high height, 

gear has internal gear to allow it to drive the axle while sliding along the axle, and external gear for the actual interfacing with other gears.

looooo, 2017-06-27:
I think the internal gear should be a new class, as there are some parameters which are not the same. It should work by creating the inner and outer wire and extrude them.

Please post a link which shows a toothed axle.

looooo, 2017-06-27:
maybe better have only one issue for this topic:
https://github.com/looooo/FCGear/issues/6

closed 2017-06-27 by looooo