looooo/freecad.gears #19: Doesn't show up in workbench menu

eimiss, 2017-10-07:
Using Arch Linux and FreeCAD 0.16. Did this:
```
$ cd ~/.FreeCAD
$ git clone https://github.com/looooo/FCGear.git
$ ln -s '/home/user/.FreeCAD/FCGear' /usr/share/freecad/Mod
$ rm *.cfg
```
It just doesn't show up in workbench menu:
[image]

looooo, 2017-10-08:
any red colored errors in console?
also you can directly add it to .FreeCAD/Mod/

dariankbrown, 2018-01-26:
I have the exact same issue

dariankbrown, 2018-01-26:
[image]

dariankbrown, 2018-01-26:
but no entry at all in the menu

looooo, 2018-01-27:
can you post the version of freecad you are using? Also have a look into the log file~/.FreeCAD/log and the output of the reportview-panel.

eimiss, 2018-02-16:
```
$ cat ~/.FreeCAD/log
cat: /home/eimis/.FreeCAD/log: No such file or directory
```

```
OS: Linux
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.16.Unknown
Build type: Release
Python version: 2.7.14
Qt version: 4.8.7
Coin version: 3.1.3
OCC version: 6.9.1
```

#Report view Panel
Run Gui init script
Init: Running FreeCADGuiInit.py start script...
Init:   Searching modules...
Init:      Initializing /usr/lib/freecad/Mod/Sketcher... done
Init:      Initializing /usr/lib/freecad/Mod/ReverseEngineering... done
Init:      Initializing /usr/lib/freecad/Mod/Image... done
Init:      Initializing /home/eimis/.FreeCAD/Mod/Curves... done
Init:      Initializing /usr/lib/freecad/Mod/PartDesign... done
Init:      Initializing /usr/lib/freecad/Mod/Ship... done
Init:      Initializing /usr/lib/freecad/Mod/Plot... done
Init:      Initializing /usr/lib/freecad/Mod/MeshPart... done
Init:      Initializing /usr/lib/freecad/Mod/OpenSCAD... done
Init:      Initializing /usr/lib/freecad/Mod/Fem... done
Init:      Initializing /usr/lib/freecad/Mod/Spreadsheet... done
Init:      Initializing /usr/lib/freecad/Mod/Draft... done
Init:      Initializing /usr/lib/freecad/Mod/Test... done
Init:      Initializing /usr/lib/freecad/Mod/Import... done
Init:      Initializing /usr/lib/freecad/Mod/Complete... done
Module: Part
Loading Part module... done
**Init:      Initializing /usr/lib/freecad/Mod/FCGear... done**
Init:      Initializing /usr/lib/freecad/Mod/Material... done
Init:      Initializing /usr/lib/freecad/Mod/Raytracing... done
Init:      Initializing /usr/lib/freecad/Mod/Mesh... done
Init:      Initializing /usr/lib/freecad/Mod/Path... done
Init:      Initializing /usr/lib/freecad/Mod/Part... done
Exploded animation workbench loaded
Init:      Initializing /home/eimis/.FreeCAD/Mod/exploded_animation... done
Init:      Initializing /usr/lib/freecad/Mod/Arch... done
Init:      Initializing /home/eimis/.FreeCAD/Mod/assembly2... done
Init:      Initializing /usr/lib/freecad/Mod/Robot... done
Init:      Initializing /usr/lib/freecad/Mod/Points... done
Init:      Initializing /usr/lib/freecad/Mod/Idf(InitGui.py not found)... ignore
Init:      Initializing /usr/lib/freecad/Mod/Inspection... done
Init:      Initializing /usr/lib/freecad/Mod/Drawing... done
Init: Loading FreeCAD GUI
Init: Running FreeCADGuiInit.py start script... done
Init: Activating default workbench StartWorkbench
Loading GUI of Part module... done
Init: Showing main window
Main window restored
Show main window
Toolbars restored
Couldn't connect to spacenav daemon
Init: Entering event loop
Init: Processing command line files

looooo, 2018-02-17:
I have to admit, that I have no idea what the problem is.

eimiss, 2018-02-17:
I guess you are not using Arch?
@dariankbrown, how about you?

looooo, 2018-02-17:
Arch linux? No I test on windows, ubuntu and centos.

johnsonm, 2018-05-06:
I'm guessing that `<None> W,9` is FCGear; that's what I see. When I bring up that workbench, though, it is totally empty. No log file, nothing in report view panel. Using FreeCAD 0.16 on Fedora 27 x86_64.

looooo, 2018-05-06:
yes, if something went wrong during startup freecad can be quite silent. But anyway, I don't support 0.16 anymore.

johnsonm, 2018-05-06:
I just downloaded the current 0.17 AppImage file from freecad.org and it also shows the same problems for me: `<None> W,9` in the workbench selector, and when that's selected, and empty user interface, with no error messages on the console, no log file...

[image]

[image]

I'm able to open the examples/bevel_gear_example.fcstd and the parameters show and the gear renders correctly.

looooo, 2018-05-07:
I see a Gear-symbol at the bottom of the list ; )

johnsonm, 2018-05-08:
Oh, looking at the wrong entry, :blush: 
Thanks!

engineeror, 2018-05-08:
It works!
`# ln -s /home/user/.config/FreeCAD/Mod/FCGear /usr/lib/freecad/Mod/`
It doesn't find it in the home dir.

Tooniis, 2020-03-16:
It doesn't appear for me on 0.18

Glasses8448, 2020-04-20:
same here: It doesn't appear for me on 0.18, kubuntu 19.10

meirm, 2020-09-19:
Doesn't show on Debian Buster Freecad 0.18

Glasses8448, 2020-09-19:
> o

Start Freecad from the terminal, by typing "freecad"
does it say "FreeCAD 0.18.4, Libs: 0.18.4R" ?
I also get the meassge "GitPython not found. Using standard download instead.", after installing the Add-On. You have to restart freecad.

But it works just fine on my machine.
In April on kubuntu 19.10. I figured out, that the repository of ubuntus apt-get was outdated and incomatible with FCgears. Try to install it via "Stable PPA through the console"

https://wiki.freecadweb.org/Install_on_Unix