looooo/freecad.gears #21: feedback from https://forum.freecadweb.org/viewtopic.php?f=24&t=27424

looooo, 2018-03-14:
- [ ] split functions (python / freecad) (make python only more useable)
- [ ] new style module
- [ ] simpler bevel-gear parameters
- [ ] spherical face for bevel-gear construction

abdullahtahiriyo, 2018-03-18:
After having played a little with bevel gears, I think it would be great:
1. Define two heights, or two "effective radius" for the start/end surfaces of the bevel gear.
2. Move the origin to the lower surface (i.e. translate the origin)

The latter is very convenient when further working on the result of the Gear WB.

looooo, 2018-03-20:
regarding 1:
this will make the module obsolete. The module for the bevel gears in FCGear is defined only for the mid-profile. Bevel-gears do not have a constant module...

regarding2:
https://github.com/looooo/FCGear/commit/a2c6f966863e603be725bdc06c402e0a97ca4ef1
this branch should be installable side by side to the master-branch:
```
pip install https://github.com/looooo/FCGear/commits/new_style_module
```

abdullahtahiriyo, 2018-03-21:
I see. Now I think I understand why it was programmed the way it was.

I have tested it.

For building on the gear, having the origin meet a face is very nice. 

I am not sure how I would want it to be, if I were to design (as opposed to just copy) a bevel gear system. 

Copying seems "easy" as, if I have understood you properly, it does not quite matter which module I use, as far as cut the gear sufficiently offset. The "module" would be irrelevant for that gear, as the mid-profile may even be cut out. Or in other words, I could get the same "piece of gear" with different module values, by cutting it at a given (different) offset position.

However, the whole point of having the origin on a face is not to have to cut it at other position, thereby demeaning the sense of having the origin there. So it seems that it is either never cutting-off that face, or having a way to boolean cut those face within, which I do understand from your text is some kind of demeaning the mid-profile module...

I will think about it a little bit more and will come back.

abdullahtahiriyo, 2018-03-22:
I have started this thread so that other that know better than me can give their views:
https://forum.freecadweb.org/viewtopic.php?f=8&t=27626#p222656