looooo/freecad.gears #22: Crown Gear, objects not cut out

4styler, 2018-08-28:
Hello Guys,
i have tried your Mod (really great work :)) and i think i have found an error.
When i build a Crown gear than the "negative teeths" are not cut out.
(Die Negativzähne wurden aus dem Zahnkranz nicht ausgeschnitten)

When i change line 329 to:
if not fp.construct:
it is working.
Can you please check it?

It was the first and is the only Mod on my newly installed FreeCAD 0.17 13528

Kind regards
4styler

looooo, 2019-01-04:
The cutting operation takes quite some time, so there is a construct property. You can simple set this property to false to get the cutted object.

bevinhex, 2019-11-08:
I suppose you already thought about increasing python performance overall as well, great job.

looooo, 2019-11-08:
the time used for the cut-operation is not limited by python but opencascade.

bevinhex, 2019-11-11:
Just a thought, I wonder if it is possible to make FreeCAD processes async, so that user can keep doing their stuff, result can show up later without freezing the interface, with that, even though the operation takes time, as long as it is indicating doing something, and showing gradually, users won't mind

closed 2020-09-22 by looooo