looooo/freecad.gears #23: Larger tooth count on involute gears causes loss of detail on teeth?

waterlubber, 2018-11-29:
I'm trying to set up a couple simple involute gears, but for whatever reason on high "tooth" counts the model drops to a lower accuracy:
[image]

FreeCAD details:
`OS: Linux Mint 18.3 Sylvia
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.18.15263 (Git)
Build type: Release
Branch: master
Hash: 7ceac311b3c5dc1b89e85b15b310ddd1d1673561
Python version: 2.7.12
Qt version: 4.8.7
Coin version: 4.0.0a
OCC version: 7.3.0
Locale: English/UnitedStates (en_US)
`

Smaller gears, around 30 teeth, work fine. What gives?

looooo, 2019-01-04:
What do you mean with "lower accuracy"?

looooo, 2019-03-11:
A gear with many teeth has similar tooth-shape as a rack. So this is not a low accuracy.

closed 2019-03-11 by looooo