looooo/freecad.gears #24: part not updated after parameters are modified

v923z, 2019-03-11:
Hi Lorenz, 
I have just installed your addon in the very latest version of freecad. I can create cogwheels, but whenever I change any of the parameters, the part is not updated. The only way of making it work is, if I insert a new cogwheel. I don't know if this behaviour is intentional (I don't believe). If not, could you, please, point out, where I have to hack the code to fix this issue? I would really like to have a somewhat prompter visual feedback.
Thanks,
Zoltán

looooo, 2019-03-11:
Hello Zoltà

difficult to say. The updating mechanism was not always the same, so it also depends on the version of FreeCAD.

For this module I used the [Part::FeaturePython](https://github.com/looooo/freecad.gears/blob/master/freecad/gears/commands.py#L50) class and the interface function for the updating mechanism is [`execute`](https://github.com/looooo/freecad.gears/blob/master/freecad/gears/features.py#L124L159) from the python proxy. Maybe there is more responsive function (similar to onChanged) for the viewprovider, but I am not sure.

So sry, I can't help much with this issue. Maybe it's best to ask in the FreeCAD-Forum.

v923z, 2019-03-12:
Thanks for the feedback! I am running the latest build of freecad (version 0.18, 16079 git). I will try to read and understand your code, and perhaps, I am able to find out what the problem is. If i succeed, I post the solution here. Otherwise, i might raise the issue in the forum, as you suggested.

evilphish, 2020-06-29:
Version 0.18.4 - No updates for me as well

RandoMan70, 2020-11-12:
The same issue.
OS: Windows 10
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.18.4 (GitTag)
Build type: Release
Branch: releases/FreeCAD-0-18
Hash: 980bf9060e28555fecd9e3462f68ca74007b70f8
Python version: 3.6.6
Qt version: 5.6.2
Coin version: 4.0.0a
OCC version: 7.3.0
Locale: Russian/Russia (ru_RU)

looooo, 2020-11-12:
I do not have such an issue. Not every change of parameter triggers the recomputation. You have to click into another parameter-field to get the parameter-change applied.

RandoMan70, 2020-11-12:
Indeed, seems it's issue of freecad 18.4. It does not update any object when changed. Updated to 19 alpha, everything works. Thank you!