looooo/freecad.gears #31: Workbench failure: something is wrong with the numpy installation

Foadsf, 2019-09-24:
My environment is 

* OS: Windows 10
* Word size of OS: 64-bit
* Word size of FreeCAD: 64-bit
* Version: 0.18.16131 (Git)
* Build type: Release
* Branch: releases/FreeCAD-0-18
* Hash: 3129ae4296e40ed20e7b3d460b86e6969acbe1c3
* Python version: 3.6.6
* Qt version: 5.6.2
* Coin version: 4.0.0a
* OCC version: 7.3.0
* Locale: English/UnitedStates (en_US)

and I have installed the package through Addon manager, I which it seems to just pull from this repository and copy into the addons folder. 

However, when I try to run the workbench I get this error:

<pre>
     [image]           
</pre>

> Workbench failure

> Somthing is wrong with the numpy instalation. While importing we detected an old version of numpy in['C:\\FreeCAD 0.18\\bin\\lib\\site-packages\\numpy']. One method of fixing this is to repeatedly uninstall numpy until none is found, then reinstall this version.

Foadsf, 2019-09-24:
Nevermind. Reinstalling FreeCAD solved the problem for some reason. I don't know what was wrong!  ¯\\\_(ツ)\_/¯

looooo, 2019-10-02:
I guess we can close this one. Numpy and freecad have to be compatible. Not an issue with freecad.gears.

closed 2019-10-02 by looooo