looooo/freecad.gears #33: [feature request] gear box

Foadsf, 2019-09-25:
I don't think that gears ever exist in solitude in any mechanism. We usually use gears in pairs of two or more to exchange torque and speed or change direction. Right now we have to create different gears separately and place them manually, which is not only very difficult but very error-prone. So it would make sense to have the option to design gearboxes in one run. For example, the workflow would be

* select the number of connections representing every meshing. for example, a gearbox of 3 gears has two connections. IT could also be considered if the gears will be in paralel ( e.g, differential joint,  planetary gear, cycloidal drive) or in series. 
* select the type of each connection
  * parallel axis (distance, gear ratio) --> two spur or helical gears
  * intersecting (angle, gear ratio) --> bevel / cone gears
  * skew axes (angle, distance, gear ratio)
* select the set controlling parameters. for example for spur gears `{m, 𝛼, w}`, or `{p, 𝛼, w}`, or `{CP, 𝛼, w}` (where `w` is the width, `p` is the reference pitch, `𝛼` is the pressure angle, and `CP` is the circular pitch using the nomenclature of [this reference](https://khkgears.net/new/gear_knowledge/abcs_of_gears-b/basic_gear_terminology_calculation.html)).

looooo, 2020-03-04:
I wonder why you get tired of pumping? or is this your job ; ).

luzpaz, 2020-03-04:
Sorry. I see it as planting seeds and hoping they grow

stefan123t, 2021-01-02:
@Foadsf, you are right gears seldom come alone! I suspect them to usually appear at least in pairs or even larger numbers.
I am just newly delving into Gear design in order to re-construct the mechanism of an medieval astronomical clock from the city hall of Ulm. This shall serve as a more complex example for what might be required from a feature request for a complete gear box.
I have setup the following initial set of gears to cover the basic functionality but I am unable to complete the gears for at least the following features missing:

https://geargenerator.com/#75,75,25,60,1,21,86261.39999998006,22,1,10,2.5,4,27,-45,0,0,0,0,0,0,80,20,4,27,-25,0,0,0,0,1,1,60,15,4,27,0,0,0,0,0,2,0,96,24,4,27,-45,0,0,0,0,1,1,38,9.5,4,27,-45,0,0,0,0,4,0,118,29.5,4,27,-45,0,0,0,0,1,1,12,3,4,27,-45,0,0,0,0,6,0,84,21,4,27,90,0,0,0,0,7,0,12,3,4,27,-30,0,0,0,0,8,0,52,13,4,27,-30,0,0,0,0,9,0,16,4,4,27,10,0,0,0,0,10,0,56,14,4,27,-50,0,0,0,0,11,1,10,2.5,4,27,-75,0,0,0,0,12,0,100,25,4,27,-80,0,0,0,0,13,1,7,3.5,2,27,-110,0,0,0,0,14,0,73,36.5,2,27,-190,0,0,0,0,4,1,38,9.5,4,27,-45,0,0,0,0,16,0,118,29.5,4,27,-45,0,0,0,0,15,1,79,19.75,4,27,-190,0,0,0,0,18,0,20,5,4,27,-190,0,0,0,0,19,1,20,5,4,27,0,0,0,0,0,20,0,75,18.75,4,27,-10,0,0,0,0,0,0,3,-1408

1. the clock contains several gears which are rotating around the same axis, but which are not locked to the same spindle.
You can see the following two schematics drawings from the following page for the general layout of the gear set.
http://www.astrouhr.telebus.de/technik/ZeigerschemaAstroUhrUlm.gif
http://www.astrouhr.telebus.de/technik/AstrogetriebeUlmSkizze_Schmitt2_1000x800.gif
It requires to align the intermediate/surrounding spindles so that the center of the axis of the main spindles driving the clock handles i.e. spindles D, E-F, W-H, G, Q, O-N are all pivoting around the same point.
It would require to modify the angle and/or radius/diameter of adjacent gears to match the necessary gaps/distances probably resulting in some float values for connection angles and/or diameter/radius of the gears.

2. allow the axis to be fixated on some other gear plate to allow so called differential gears.
I understand that the three dimensional driving gears of the clock are using bevel gears, which is strictly off limits for the 2D layout algorithm implemented at geargenerator. But apart from that it also includes a so called differential gear in 2D (see axis P and L-M, which both are fixated on the main gear G and are driving together - in different directions - the two final axis Q and O-N).

3. There are some worm gears that only have one tooth per rotation according to the first diagram above.
I still have to check the live model being exhibited in the city of Ulm townhall for the actual driving gear of S-T, U, V, W-H.
The photographs from the student model displayed on the above homepage may not be detailled enough to understand the mechanism.
http://www.astrouhr.telebus.de/modell/modell.html

4. for an easier matching between the drawing schematics given on the above pages I would like to give each gear, spindle or axis a descriptive name.
This would be both a name (e.g. simply the letters given in the schematics) plus some possibly longer description what the respective handle should be showing.

5. the geargenerator.com adds a _label_ for each gear, including the gear #, number of teeth N, pitch diameter D, diametral pitch P and pressure angle PA plus some gear _guide lines_ including the pitch and other diameter guide lines and the principal tooth indicating the connection surface with the adjacent gear.

6. it allows animation of the gear box by simulating the rotation of the first gear and rotating the connected gears likewise.

Maybe it would be sufficient to incorporate a 2D spur gear layout algorithm first and add 3D bevel / worm gears later maybe even with another separate feature / issue request.