looooo/freecad.gears #34: Rack wrong dedendum

schwantz69, 2020-02-03:
Hi,
I'm new here (and in FreeCAD too :-) ). I'm using FCGear because I need to create a Rack. 
I have test FCGear but I found a problem:
In my sample I have try to creatre a rack 24x24 Module 2 but after I have create it I found that the teeth are wrong in fact the teeth addendum and the teeth dedendum are always equal (in my example was alway 2 but the correct is Addendum = 2 and dedendum = 2,5).
How I can solve this important topic ?
Can you please add the possibility to indicate this two paramenters ? It could be useful also in case of teeth modification.
Thank you for your precious work !

looooo, 2020-02-03:
I guess we should add a `clearence` parameter. But for now I simple added your suggestion:
https://github.com/looooo/freecad.gears/commit/b2d8fa7fbf036da3df5fe3eee459cad3d01d1690

To add the parameter I would like to find a good way to keep backward compatibility first. So let's keep this issue open.

[image]

schwantz69, 2020-02-03:
> b2d8fa7

Dear Lorenz,
Thank you very much for your fast reply !
I see your modification in py module but... how I can use it ? I'm sorry
but I'm new on FreeCAD and I guess the py module modification is for
advanced user ! 😀
So please can you send me detailed instructions ?
Thank you very much, your help is precious for me.

p.s. many compliments for your work !

looooo, 2020-02-03:
> how I can use it ?

Assuming you installed freecad.gears (FCGears) with the addon-manager, you can update the latest changes with the same tool. https://www.freecadweb.org/wiki/Std_AddonMgr

schwantz69, 2020-02-03:
Oh, sorry.... 
Ok I have test it but unfortunately I'm was not clear and I'm very sorry.
As you for sure know there is different type of teeth profile under DIN 3962 - 3967 normative.
The most used is accordin DIN 3962 Profile II, this normative define also the position of the pitch line in the teeth height where:
Addendum = Normal module
Dedendum (module + cleareance) = Normal module * 1,25
so it mean that for easely calculate the total teeth deep is sufficient moltiply the normal module x 2,25 in fact:
Module 2 = 2*2,25 = 4,5mm = (2 + (2*1,25))
Module 3 = 3*2,25 = 6,75mm = (3 + (3*1,25))
.......

Could you please modify your script with the described formula ? Thank you !
And excuse me another time if in my previous post I'm was not clear.

looooo, 2020-02-03:
> As you for sure know there is different type of teeth profile under DIN 3962 - 3967 normative.

Do you have a drawing? I do not really understand what is not correct with the recent changes.
With the addition of the clearence the total tooth-height can be calculated like this:
`hight = ((1 + clearence) + (1 + head)) x module`
So for head = 0, clearence = 0.25 and module = 2 you will get a total height of  `2 x ((1 + 0.25) + (1 + 0) = 4,5`. Isn't that correct?

[image]

schwantz69, 2020-02-04:
Dear Lorenz, 
You are right ! Probably something was wrong during the update because this morning I have try to replicate the problem but is correct ! I'm very sorry.
Unfortunately I have another issue.....
I have built a rack 29x29 Module 3 helical teeth 20° (see the attachment), as for checking the teeth I use a roller Diam. 6mm and the roller size should be 31,297 but is 31,222.
I have check and the problem is the pressure angle that should be 20° but in realty is 21,1728°.
I have made a mistake ?
Thank you for your help
[image]
[image]

looooo, 2020-02-04:
The gear-properties are internally recomputed to match a rotated tool with the user-given properties. So module and alpha are recomputed.
I added a property (tool_properties) which you can set to `False`:
[image]

The same is now possible with the involute-gear:
https://github.com/looooo/freecad.gears/commit/397c10fdf23081342c651319523ba88210c7b238

Hopefully this solves your issue.

schwantz69, 2020-02-04:
Dear Lorenz,
Unfortunately no... take a look below
Thank you
[image]
[image]

looooo, 2020-02-04:
How is he angle measured?

looooo, 2020-02-04:
Ah I see now the problem. There is indeed something wrong with the internal computation of alpha.

schwantz69, 2020-02-04:
> 
> 
> How is the angle measured?
> Is the sketch placed on the fron-face of the gear or are you cutting the gear by a plane rotated by beta?

Rotated plane

schwantz69, 2020-02-04:
> 
> 
> Ah I see now the problem. There is indeed something wrong with the internal computation of alpha.

Great !

looooo, 2020-02-04:
ok, it should be fixed. Please try again.

schwantz69, 2020-02-04:
Dear Lorenz,
I'm sorry no.... 
[image]

looooo, 2020-02-04:
ah sry, you need to enable properties_from_tool again.

[image]

schwantz69, 2020-02-04:
Perfect ! Now work very well.
Thank you very much for your help !

closed 2020-02-04 by schwantz69