looooo/freecad.gears #35: Rack Lenght

schwantz69, 2020-02-05:
Dear Lorenz,
Unfortunately I have found another small issue that I hope is very easy for you to solve.
The issue is regarding the total lenght of the rack, on the attached pictures you can see I have defined a rack 29x29 Module 3 with 100 teeth helical teeth 19,528333 Deg. Angle that mean a transversal Module 3,18310 that correspond to pitch 10,0000mm
With the mentioned data the theorical lenght of the result rack it will be 1000mm (we assume that both heads start exactly with half teeth gap).
Unfortunately the result lenght of the rack is 997,898mm, it seems because the rack do not start end end in the middle of teeth gap.
Do you think there is the possibility to solve this issue easely ?
Thank You !
[image]
[image]

looooo, 2020-02-05:
can't reproduce:
[image]

To get 100 x pitch you need to create 101 teeth and measure at the same points of the profile. (see my attached picture)

schwantz69, 2020-02-05:
Dear Lorenz,
Unfortunately no, all the racks are measured like my DWG.
For example look at the attached document pick from a catalogue of a famous German producer.
[image]
[image]

looooo, 2020-02-05:
Ok, making the ending like described in your drawing make sense. I changed that:
https://github.com/looooo/freecad.gears/commit/95f6de16fb27148074163d9b4c7bb3ab8d116ee7

schwantz69, 2020-02-05:
Dear Lorenz,
Thank you ! I have tested and it is ok.
If you have in plan to visit Italy tell me, it would be a pleasure to meet you and of course drink a beer together !

closed 2020-02-05 by schwantz69