looooo/freecad.gears #41: Failure to show in menu's in FreeCAD 0.18.0

stephenos1phillips, 2020-03-19:
Debian install (thus not latest).
A [pastebin link of FreeCAD.log ](https://pastebin.com/6yqPCm8Q)
FCgears does not appear in the workbench list but the Addon Manager states its installed.
The only signs are Init.py and InitGui.py are both missing from .FreeCAD/Mod/FCgear (which is ignored).
I tried freenode #freecad but .. anyhow since the last issue closed, another added to that issue 3 days ago they couldn't get "gears" in the menu. I've reinstalled numerous times with the same results.
For some reason I get " Something went really wrong, and we can’t process that file. " when I attempt to attach an image of the menu missing 'the gear'.

looooo, 2020-03-19:
Please use py2 Branche of freecad.gears. Latest master needs a more recent freecad.

stephenos1phillips, 2020-03-20:
hmm OK
I tried 
`pip install git+git://github.com/looooo/freecad.gears@py2`
since I can't use FreeCAD to install a branch, however this did not work, and it didn't appear in the menu system.
I don't believe I can use pip install https://github.com/looooo/FCGear/archive/master.tar.gz since that would install the master (python3). So I tried the following next to be certain python 2 pip is used:
first clean up to remove what was installed
`pip uninstall freecad.gears`
then install using
`pip2 install https://github.com/looooo/FCGear/archive/py2.tar.gz`
just to be certain whatever was installed was removed.
After executing that, the directory FCgear did not appear in path
`.FreeCAD/Mod/`
I'm a bit baffled. I am not sure I should build FreeCAD 0.18.4 but I suppose I can do that like I use to in Debian.
I'll also message the Debian buster backports (a bit of alliteration there) list that it might be good to update FreeCAD as it has undergone 4 revisions and a year since 0.18 was added to buster last did so, the last revision being in December 2019.
As an aside, is it possible during install to detect the version of FreeCAD and notify the user that a new version (of FreeCAD) is needed? 
I'm still wondering what's going on as I can write python code but the install system is something completely different (LOL). Thanks for the help thus far.

looooo, 2020-03-20:
> After checking Debian backports for udpates to FreeCAD it states "freecad is already the newest version (0.18~pre1+dfsg1-5)" likely this means that they haven't updated freecad in the backports to 0.18.<anything greater then 0>. I can message backports and see what they say.
As an aside, is it possible during install to detect the version of FreeCAD and notify the user that a new version (of FreeCAD) is needed?

Just a short comment:
Please try the 0.19 appimage here:
https://github.com/FreeCAD/FreeCAD/releases/tag/0.19_pre  > assets > *.appimage

looooo, 2020-07-17:
there is some confusion with this, but actually this should not be an issue. best to use a 0.19 prerelease.

closed 2020-07-17 by looooo

hanelyp, 2021-08-22:
Using freecad 0.19, debian install on linux mint.  Still seeing the exact same problem with the fcgear workbench.