looooo/freecad.gears #42: [feature request] Migrating FCgear and the future FCgearbox to assembly4

kcleung, 2020-03-27:
Given the development of the assembly4 addon, which adds assembly support to FreeCAD in a low cost way, it will make sense to migrate FCgear and the future FCgearbox to use assembly4.

I guess Assembly4 will also facilitate our implemetation of FCgearbox.

Is it a good time to make a fork, where FCgear and FCgearbox will be implemented as assembly?

This fork will depend on Assembly4.

looooo, 2020-03-29:
I am not sure if asm4 is the way to proceed but if you think so, feel free to share your thought on how to do it. I don't have time for such things right now.

stefan123t, 2021-02-04:
Here is a step by step video from MindScraper detailling the steps to construct a gearbox usign FCGear and Assembly 4.
* It shows how to parametrize the cog wheels
* as well as the shafts added to the gears
* and then takes the critical step of constructing the diametral pitch circle
* then assembly4 is used to describe the layout of the pieces and their dependencies
* it even creates the gearbox side walls with bearings (maybe add some slack)  
https://www.youtube.com/watch?v=zYXXfDFg6LY

I think the previous video did not show the calculation to turn the second wheel for the angle of 1/2 teeth (180/teeth) to make them align with interlocking teeth.
This is shown in part 2/6 of Evgeniy Ivanov tutorial using Involute Gear (not FCGear *sic*)
https://www.youtube.com/watch?v=nMhDR3_oeEc

Another honorable mention is the following two videos from Jacques Favre showing how to construct it with FCGear and assembly 4 workbench and in the second part even animate it.
https://www.youtube.com/watch?v=-4qlmS25Mk8
https://www.youtube.com/watch?v=eLTrfCl2OG4

@kcleung how would you implement such an FCgearbox plugin / workbench making use of both FCGear and Assembly 4 and generating a set of gears and the matching constraints ?

jacprog, 2021-02-17:
Hi 
thanks for including me here
I made this clock:
https://forum.freecadweb.org/viewtopic.php?f=24&t=55567
7 pair of gears in XY plan in 6 Z plans, not sure any assembly could help
 I think the gear workbench is great as is