looooo/freecad.gears #43: Unable to create an Involute Rack

antoinejaunard, 2020-04-08:
Hi,

I'm using FreeCAD 0.19 on Manjaro-Gnome with Python 3.8.2 installed.

I installed the Gear workbench using the add-on manager. Everything seems to work properly, except the creation of a rack.

I've got this message in the console:

```
Traceback (most recent call last):
  File "/home/antoine/.FreeCAD/Mod/FCGear/freecad/gears/features.py", line 249, in execute
    pts = fp.rack.points()
  File "/home/antoine/.FreeCAD/Mod/FCGear/pygears/involute_tooth.py", line 219, in points
    teeth = [ext1.tolist(), ext1.tolist()] + teeth + [ext2.tolist(), ext2.tolist()]
<class 'ValueError'>: operands could not be broadcast together with shapes (2,2) (60,2) 
8.94e-07 <App> Document.cpp(3715): Failed to recompute gears_00#InvoluteRack: operands could not be broadcast together with shapes (2,2) (60,2) 
Recompute failed! Please check report view.
```
Any idea on what to do?

antoinejaunard, 2020-04-08:
My bad, I had to check the `add_endings` to `fase` in order to get it.
Maybe add it as a default option?

Thanks, have a nice day.

closed 2020-04-08 by antoinejaunard

looooo, 2020-04-08:
https://github.com/looooo/freecad.gears/commit/8193f1ebcebe5edb36fe6219540a56a90def2bde#diff-2f6371df3e32bdc716876a1a29441df4R221