looooo/freecad.gears #44: switch license: gpl-3

looooo, 2020-04-09:
@luzpaz @kcleung I would like to move to gpl3. As you contributed to this library, I would like to know if this is fine for you.

kcleung, 2020-04-25:
Hey looooo

Apart from a standalone plugin, the freecad.gear backend can also be used as a library for some other applications.

obviously, the intention is that any derivative of freecad.gear must also be open source.  However, if it is licensed to the vanilla GPL, (such as v 3), then other software / plugin that merely calls the function of freecad.gear must also be GPL then This will hamper the use of freecad.gear

So perhaps the lesser GPL (LGPL) 3 will be more appropriate.  This will enable other software or plugins with other licenses (such as proprietary) that merely call the function of freecad.gear, but does not use or incorporate any freecad.gear code, to call functions of freecad.gear

looooo, 2020-04-25:
@kcleung i would like toll avoid lgpl. Because using lgpl allows anyone to fork the module and offer a commercial alternitive instead of contributing.

looooo, 2020-05-04:
@kcleung gpl3 vs gpl2 is not a big deal. Please give your ok, so we can move on.

kcleung, 2020-06-26:
> 
> 
> @kcleung gpl3 vs gpl2 is not a big deal. Please give your ok, so we can move on.

I give consent to switch to GPL3

closed 2020-06-26 by looooo