looooo/freecad.gears #45: Impossible to install/access to FCGear Workbench

Banane84, 2020-05-09:
Hello I'm using FreeCad 0.18~pre1+dfsg1-5

I install FCGear workbench from addon manager. 
After restarting FreeCad no Gears workbench apears ;(

I cheched in~/.freecad/Mod I've a directory FCGears full
No error message at startup
in log grep FCGear I found
`Log: Init:      Initializing /home/x/.FreeCAD/Mod/FCGear(Init.py not found)... ignore
Log:    /home/erwand/.FreeCAD/Mod/FCGear
Log: Init:      Initializing /home/x/.FreeCAD/Mod/FCGear(InitGui.py not found)... ignore
`
I feel install just did'nt work properly
`tree ./.FreeCAD/Mod/FCGear/
./.FreeCAD/Mod/FCGear/
├── docs
│   └── crown_gear.pdf
├── examples
│   ├── animated_spiral.gif
│   ├── animation.py
│   ├── bevel_gear_animation.fcstd
│   ├── bevel_gear_example.fcstd
│   ├── gear_from_picture.fcstd
│   └── spiral.png
├── freecad
│   └── gears
│       ├── commands.py
│       ├── features.py
│       ├── icons
│       │   ├── bevelgear.svg
│       │   ├── crowngear.svg
│       │   ├── cycloidegear.svg
│       │   ├── gearworkbench.svg
│       │   ├── involutegear.svg
│       │   ├── involuterack.svg
│       │   └── wormgear.svg
│       ├── init_gui.py
│       └── __init__.py
├── LICENSE
├── MANIFEST.in
├── pygears
│   ├── bevel_tooth.py
│   ├── computation.py
│   ├── cycloide_tooth.py
│   ├── _functions.py
│   ├── __init__.py
│   ├── involute_tooth.py
│   └── profile.py
├── README.md
└── setup.py
`

looooo, 2020-05-10:
Best to use the py2 branch for 0.18. most likely the wb is hidden by a freecad/__init__.py file. There should exist only one such file.
The py2 branch is old-style and should work out of the box with freecad 0.18.

Banane84, 2020-05-10:
Thanks it worked directly

closed 2020-06-23 by looooo