looooo/freecad.gears #46: Problem with module in helical rack

LeeKC0001, 2020-05-10:
I am completely new to FreeCAD and other CAD. In a project to re-create a helical rack by 3D printing, I measured the distance between 3 teeth of a rack and get a value of 4 mm.

So I calculate the module by this formula (4 mm / 3 / pi).

However, the distance between 3 teeth generated by FCgear workbench is 4.26 mm.

[image]

I noticed that the distance between 3 teeth is 4 mm in a spur rack (beta set to 0).

[image]

Did I missed out something?

Your help is much appreciated.

Best regards.

KC

looooo, 2020-05-10:
You can switch properties_from_tool to false. Then the input module is used as the traverse module. I guess this is what you want.

LeeKC0001, 2020-05-10:
I see. Thanks a lot for clearing up my misconception!

closed 2020-05-10 by LeeKC0001