looooo/freecad.gears #49: Bevel gears wrong angle?

Schildkroet, 2020-07-10:
Hello,
i was playing around with Gear-WB and it seems that the angle of bevel gears is incorrect. I created to identical bevel gears with following params:
[image]

If i place them together with 90° degree, the angle of the teeth is not matching:
[image]

looooo, 2020-07-11:
please set "reset_origin" to false and do not add linear translation (only angle).

looooo, 2020-07-17:
btw. the faces of the tooth-heads should not be parallel. The gears are not correctly placed.

closed 2020-07-17 by looooo