looooo/freecad.gears #50: add option to create fillets

looooo, 2020-08-18:
fillets instead of undercut and at the top of the teeth would be a nice addition.

looooo, 2021-06-03:
ref.: https://forum.freecadweb.org/viewtopic.php?f=22&t=49562

looooo, 2021-06-04:
using pyocct as long as there is no freecad function for this:

```
import numpy as np
import FreeCAD as App
import Part
from OCC.Core import ChFi2d
from OCC import Core

api = ChFi2d.ChFi2d_FilletAPI()

npt = None

def pyocc_2d_fillet(e1, e2, r):
    # asuming edges are in a plane
    # extracting vertices
    p1 = np.array([*e1.valueAt(e1.FirstParameter)])
    p2 = np.array([*e1.valueAt(e1.LastParameter)])
    p3 = np.array([*e2.valueAt(e2.FirstParameter)])
    p4 = np.array([*e2.valueAt(e2.LastParameter)])
    t1 = p2 - p1
    t2 = p4 - p3
    n = np.cross(t1, t2)
    pln = Core.gp.gp_Pln(Core.gp.gp_Pnt(*p1), Core.gp.gp_Dir(*n))
    occ_e1 = Core.TopoDS.topods_Edge(Part.__toPythonOCC__(e1))
    occ_e2 = Core.TopoDS.topods_Edge(Part.__toPythonOCC__(e2))
    api.Init(occ_e1, occ_e2, pln)
    if api.Perform(r) > 0:
    	occ_p0 = Core.gp.gp_Pnt(0,0,0)
    	occ_arc = api.Result(occ_p0, occ_e1, occ_e2)
    	global npt
    	npt = occ_p0
    	return Part.Wire([Part.__fromPythonOCC__(occ_e1),
                      Part.__fromPythonOCC__(occ_arc),
                      Part.__fromPythonOCC__(occ_e2)])
    else:
    	return Part.Shape

def test():
    App.newDocument()
    radius = 0.1
    p0 = App.Vector(1,0,0)
    p1 = App.Vector(2,0.5,0)
    p2 = App.Vector(1,1,0)
    p3 = App.Vector(1,0,0)
    e1 = Part.LineSegment(p1, p0).toShape()
    e2 = Part.LineSegment(p3, p2).toShape()
    Part.show(e1)
    Part.show(e2)
    Part.show(pyocc_2d_fillet(e1, e2, radius))

test()
FreeCAD.Console.PrintMessage(npt.Coord())

```

looooo, 2021-06-04:
this is now possible with this branch;
https://github.com/looooo/freecad.gears/tree/megagrants

parameters:
- head_fillet
- foot_fillet

Both have no dimension and are used to compute the fillet radius by multiplying with the module.

[image]

rockstorm101, 2021-06-14:
First of all thanks for the workbench. This is great. I've been testing it and this tip/root radius (which is how it is called in the standards usually) is the first feature I was missing. Looking forward to see it implemented. Thanks again.

jbaehr, 2021-11-04:
I just tried it on the develop branch. cool stuff!

I've one addition/enhancement to propose: What do you think about two additional (hidden?) computed properties `max_{head,root}_fillet`, with the maximum possible value for the fillet? The radius may not be larger than half the traverse-length of the bottom/top land, but this value again depends on profile shift, number of teeth, clearance/head, maybe more.
This would allow to use e.g. `=min(clearance, max_root_fillet)` as value for the root_fillet.

jbaehr, 2021-11-04:
> ... tip/root radius (which is how it is called in the standards usually)

Which standard are you referring to? ISO 1122-1:1998 "Vocabulary of gear terms" (publicly available as [Indian Standard 2458:2001](https://law.resource.org/pub/in/bis/S01/is.2458.2001.pdf)) calls what FCGear calls "root_fillet" just "fillet" and there is no equivalent for the tip defined.

In addition, there is at least a Japanese and an American standard out there, too. In the [technical docs from QTC Gears](https://qtcgears.com/tools/catalogs/PDF_Q420/Tech.pdf), which symbol names are used within FCGear, different standards are compared. The document itself uses both terms, "Tip/Root Radius" and "root fillet".

Unless you're making a gear cutter, the tip radius/fillet is barely mentioned. Instead, some chamfer at the tip, called "semi topping" seems way more popular.

rockstorm101, 2021-11-23:
> Which standard are you referring to? ISO 1122-1:1998 "Vocabulary of gear terms" (publicly available as [Indian Standard 2458:2001](https://law.resource.org/pub/in/bis/S01/is.2458.2001.pdf)) calls [...]

Well, good point, thank you for questioning that. I'm used to that terminology at work because that's how it is called on the KISSsoft software. However, I went ahead and actually had a look at some standards I had at hand (DIN 3972, ISO 868 and DIN 1825) and there doesn't seem to be a consistent name for these. The root/fillet/radius seems to agree but the one at tooth tip is not even mentioned. Thinking about it one might wrongly assume root/tip radius to be half of the root/tip diameter so it is probably better to name those something less misleading.

> In addition, there is at least a Japanese and an American standard out there, too. In the [technical docs from QTC Gears](https://qtcgears.com/tools/catalogs/PDF_Q420/Tech.pdf), which symbol names [...]

Totally agree, tip radius (or tip fillet or whatever) is rare in "traditional" gear manufacturing), only defined when topping cutting tools have such a radius at the root of the cutting tool. A chamfer/relief is (in what I've seen so far) more common.