looooo/freecad.gears #52: Getting Shape objects directly

Equidamoid, 2020-09-01:
How about making a way to get gears as Shapes without all that FeaturePython complexity?

Something like
```
def make_gear(teeth: int, module: float, ....) -> Shape:
    # most of the current InvoluteGear.execute goes here

    shape.dw = ... # can't think of a nice way of returning gear properties, maybe subclass Shape
    return shape
```

This should make it much easier to use the library from python code (less boilerplate code, type hints, etc.).

looooo, 2020-09-01:
Good idea.

Equidamoid, 2020-09-02:
I hacked together https://github.com/looooo/freecad.gears/compare/master...Equidamoid:feat/make_shapes
@looooo  what do you think?

I still want to replace exact default values with `None`. And even after that it feels like too much copypaste, but I can't figure out a nicer way of doing it.

looooo, 2020-09-02:
Yes, I think it's good. Maybe we can make the function behave the same way as the gui-command and make give defaults for all parameters (default gear: module=1, teeth=15, height=5)

regarding copy-paste: I think this is not a big issue.

This way we can also remove the `App::PropertyPythonObject` which is now unused.

Equidamoid, 2020-09-02:
I'm not so sure about having default module-teeth-height, but also don't care that much, can add them =)

Copypaste... IDK, it hurts even to write it like that. I'll try to make it better... Will it be a big deal to drop support of python older than 3.6?

Equidamoid, 2021-07-12:
@looooo 
Finally, after almost a year(!) I've got an idea that I like.
I made a small PoC with LanternGear: https://github.com/looooo/freecad.gears/compare/master...Equidamoid:hybride?expand=1

Effectively I split the LanternGear class into 3:
 - `LanternGearProperties` -- a dataclass containing all the info about properties and providing a type-safe way of accessing them from the code.
 - `LanternGearGenerator` -- the part that actually does the work, same as the old `execute`, but returns the shape instead of assigning it
 - `LanternGear` -- declaration binding the two together 

Also there is `HybrideBaseGear`, which does some chores and is supposed to be the new `BaseGear`.

As the result:
 - GUI stays the same
 - the property declaration is much shorter
 - making gear shapes in code is trivial:
```
from freecad.gears.features import LanternGearProperties, LanternGearGenerator
props = LanternGearProperties()
props.teeth = 50
gear_shape = LanternGearGenerator(props).generate_shape()
```

What do you think?