looooo/freecad.gears #55: Undercut option creates artifacts with high teeth count and high numpoints

herrgerd, 2020-09-27:
Creating a 150 teeth module 2 involute gear showed strange errors when I imported the STEP file into other CAD systems.
On further inspection I saw strange "spikes" in the model:

[image]
[image]

They only occur when using high numpoint precision (above 10) and high teeth count. Those are not **sane** settings someone would use. Just wanted to let you know!

looooo, 2020-09-27:
while undercut intersection with involute can be improved, I don't think it's a big issue for high number of teeth. 
an additional root fillet is on the todo list if this is what you want.

looooo, 2021-07-21:
Additional root fillet was added to the megagrant-branch. Please use this property and not undercut if you have a high number of teeth.

closed 2021-07-21 by looooo