looooo/freecad.gears #57: Crown gear wrong if "construct=true"

MSoegtropIMC, 2020-10-10:
If I create a crown gear with "construct=true", which appears to be the default, it looks like a boolean operation is missing. I get a zylindric ring and objects which should be subtracted from the ring to create the gear. This happens also to the default crown gear created when clicking on the crown gear button in FreeCAD 0.18.4.

If I set "construct=false", the gear looks fine.

Thanks for this great tool!

FreeCAD version:
```
OS: Windows 10
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.18.4 (GitTag)
Build type: Release
Branch: releases/FreeCAD-0-18
Hash: 980bf9060e28555fecd9e3462f68ca74007b70f8
Python version: 3.6.6
Qt version: 5.6.2
Coin version: 4.0.0a
OCC version: 7.3.0
Locale: English/Germany (en_DE)
```

FCGear version: installed today with "Addon Manager".

looooo, 2020-10-13:
This property (construct) was added because the boolean operation is quite time-consuming. It's not ideal. If you have a better idea, please share.

MSoegtropIMC, 2020-10-13:
It is not clear to me what to do with the resulting object. It looks like two parts to which I need to apply a boolean diff, but since it is only one object I don't know how.

In case this is intended behavior, it definitely needs more description. I would suggest to make the "construct=false" variant default and add a comment and instructions on how to work with the "construct=true" variant in case one needs this.

I added a screen shot how a freshly created crown gear looks in 0.18.4 and 0.19 (it is the same):

[image]

looooo, 2020-10-13:
Normally the computation of the gear is fast. So you click a symbol and change the parameters and check the changes visually. If we set the construct-property to false, you have to wait some seconds until the computation has finished. This way you are not able to observe the correlation between properties and geometry easily.

Actually there is a text printed which indicates that you should switch the construct property to get the final result.

`14:09:12  Gear module: Crown gear created, construct mode = true for improved performance. Set construct property to false when ready to cut teeth.`

MSoegtropIMC, 2020-10-13:
I see - that's the way it is supposed to work - one uses construction mode for parameter editing and then switches it off when finished with editing. This is not so clear from the name. I assumed this mode was intended to produce a further editable, more flexible object - say where I can remove some teeth if I only need a partial gear. It would be clearer if this parameter would be called "editing-preview".

While this might optimize things for a few experts, I would assume that most people spend far more time figuring out what is going on than it saves them. I usually don't have the message window open so unless this is shown in the status bar I won't see the message.

I would suggest to swap this - make construction=false the default and e.g. put in the description of the parameter or the status bar a text indicating that switching this mode on allows for faster parameter editing, but I think just changing the parameter name would also have saved me some pondering over this.

looooo, 2020-10-13:
I renamed the property. It's now called `preview_mode`. Not sure about the default value. I do not change this for now as I don't see the improvements in doing so. Feel free to close this issue.

desttinghim, 2021-03-09:
The freecad wiki is out of date on `construct` vs `preview_mode`. This would be useful info to have listed there. https://wiki.freecadweb.org/FCGear_CrownGear/en