looooo/freecad.gears #58: Involute gear thooth profile for use with worm gear

DickyOZ, 2020-10-27:
Hi, I have just downloaded this very useful gears mod. I have two issues I can't solve.
1. How do I alter the tooth profile to cater for spur gears driven by a worm drive?
2. How do calculate the distance between centres of the worm gear and spur gear for them to mesh properly?

looooo, 2020-10-27:
If you mean by 1. something like this:
[image]
such gears are not yet available within freecad.gears. 

2. The involute gear has a property `dw` which is the computed pitch diameter.

DickyOZ, 2020-10-27:
Many thanks for the prompt reply it has helped a lot.
I solved the first issue by using a torus solid to cut away the involute gear to the shape shown in your reply. At least it is close enough to match the gears I'm using.
The axial distance precision is causing me a lot of grief in 3D printing - probably a limitation of the 3D accuracy.
Many thanks for you excellent Gears module it has made my life much easier
[image]
. Incidently the latest FreeCAD (1.8) addons manager doesn't look anything like the previous version. I've programmed using many languages but have yet to dabble in python. I've attached a pic of the (exposed) gears of my home made star tracker.

looooo, 2020-10-27:
this looks impressive. If you have some time feel free to post this in the freeecad show-case forum:
https://forum.freecadweb.org/viewforum.php?f=24

looooo, 2020-10-27:
> Incidently the latest FreeCAD (1.8) addons manager doesn't look anything like the previous version.

I repaired the addon-manager for 0.18 versions (linux/mac). If you are ùsing one of these plattforms simple download the `rev1` file here:
https://github.com/FreeCAD/FreeCAD/releases/tag/0.18.4

Not sure if windows was updated yet.

DickyOZ, 2020-10-27:
Thanks looooo, the WIN 64 version hasn't been updated yet. I'll keep checking.
The Star tracker uses a Nema 17 stepper to drive the gear chain. And 28PYJ-48 steppers to align and tilt to locate the celestial pole (necessary to find the CP in the southern hemisphere). The whole thing is controlled by an ESP-32 MCU which has two CPUs  and four built-in timers. I dedicate one CPU and one very accurate timer to drive the rotation stepper.  The other CPU is used for communication and control processes. The MCU reads orientation data from an MPU-92.65 sensor. It also connects to my phone via Bluetooth to give a visual indication of the setup and to control the tracker. The phone App I wrote uses GPS to lookup magnetic declination and align the tracker. The rotation is accurate to within a few seconds a day (overkill) but the compass and accelerometer leave a bit to be desired; hence the 'Nudge' buttons.
[image]
[image]

closed 2020-10-27 by DickyOZ