looooo/freecad.gears #63: Unable to install on FreeCad 0.18

jf1452, 2020-12-16:
I cannot find freecad.gears on Addon Manager, I tried using PIP to install which seemed to complete successfully, but after running FreeCad and going into Part Design; I am unable to find the newly installed Gear tool. Any advice please?

iotricity, 2020-12-23:
Having the same problems after reinstalling Windows 10 Pro. Installed FreeCAD 0.18 rev. 4 (10/22/2019). The macro is not available in the list. Tried installing manually as described in the Wiki, restarted FreeCAD, checked if the workbench wasn't disabled. The workbench doesn't show up.

stefan123t, 2021-01-02:
I could install it using the recipe given in the following feedback thread for FCGears:
https://forum.freecadweb.org/viewtopic.php?f=8&t=27626&start=30#p332138

i.e. on my Linux box:
cd .Freecad/Mod/
wget https://github.com/looooo/FCGear/archive/master.zip
unzip master.zip
# wget will unpack into .Freecad/Mod/freecad.gears-master/ 
git clone https://github.com/looooo/freecad.gears.git
# while git clone resulted in .Freecad/Mod/FCGears/
# but both worked equally well

After (re)starting freecad 0.18.4-2019-11-10, I could use File > New, View > Workbench > Gear W, 6 to activate the FCGear Workbench and create Gears. Thanks for providing the precious parts.

jf1452, 2021-01-03:
Thank you stefan123t; I used the same approach on Windows 10 and it now working; thank you.

looooo, 2021-01-05:
There will be a freecad 0.18.5 release which should target the addon manager issue. But not sure what the status of the 0.18.5 release is  @sgrogan.

stefan123t, 2021-01-06:
Dear Lorenz @looooo, thanks for the clarification. I have tried to apply the patch on the AddonManager.py as suggested in the forum thread with no success. But since then I also installed the AppImage for 18.5 which allows installing the FCGears workbench from the Addons Dialog as described.
Kind regards, Stefan

GeoffGMartin, 2021-01-29:
Thanks for this! I had the same problem on a Mac (OS 11.1 Big Sur - which can't run FreeCad 0.19, unfortunately...)

Some minor variations were necessary to get it running on my machine (in case it helps).

- downloaded the file at https://github.com/looooo/FCGear/archive/master.zip via a browser
- expanded the .zip in my Documents directory
- Go to the FreeCad.app in the Applications folder, right-click or Ctrl-click and select "Open Package Contents" This will open a directory called "Contents"
- Move the directory called "freecad.gears-master" from Downloads to
Contents/Resources/Mod
- Close the window.
- Boot FreeCad, create a new file, and go to View > Workbench > Gear W, 6, as described by stefan123t, above.

Thanks again!

jbaehr, 2021-11-02:
> It has a version check in the code (features.py line 119). It checks for >= 19...

Yes, because the API changed there [ref](https://forum.freecadweb.org/viewtopic.php?f=10&t=54370&start=10#p513786). This is unrelated to not being able to install. An error in this check would result in a runtime error when trying to create a gear object, like #84.