looooo/freecad.gears #65: Errors/Corrections of the Undercut Option

JohnThinger, 2021-03-05:
First of all: FCGear is a great workbench!
However, I think the undercut option needs corrections.

As reported in #55, it creates "artefacts" when creating involute gears with a higher number of gears, i.e. the end point of the untercut spline does not match the endpoint of the involute curve. One could argue that you do not need undercuts when using high-number-of-teeth-gears, but in practical applications this is not really true. The situation can be _improved_ by increasing numpoints, but the result is never perfect. This leads to problems if one tries to chamfer the top of bottom of a gear as the the overall path is not closed.

Furthermore, the undercut seems to be non-perfect anyway: Even with the standard 15 teeth gear, the foot radius of the gear is not identical in the versions with undercut and without undercut.

The standard undercuts should be 

- either to quarter circles starting from the involute and ending in the foot radius 
- or half a circle from involute to involute _touching_  the foot radius.

The current solution looks more like an ellipse than circle segments.
I would really appreciate an improvement here!

[image]

looooo, 2021-03-05:
Undercut is produced by a cutting rack. That's how it is computed for the involute gears:

[image]

But an alternative fillet is on the todo list for a long time. Thanks for the information. I will try to incorporate it once I have time to work on this.

JohnThinger, 2021-03-08:
You are right; I have confused the undercut option with a generalized option for foot rounding (by either two quarter circles or half a circle). The current option creates a "true" undercut which explaines strange results when applying it to gears with a higher number of teeth. A more generalized foot rounding function would be greatly appreciated.

closed 2021-03-08 by JohnThinger