looooo/freecad.gears #66: Please add more timing gear profiles to FCGear

azalanono, 2021-03-09:
Hello, looooo.

I was very surprised to fine timing gears in FCGear workbench after doing an update. (I also upgraded to FC 0.19). I was waiting so long for a way of creating timing belt gears directly in FC.

But having a closer look, I see only "GT" belt profiles.   Could you, please, add "T" profiles? They differ a lot and are more suitable for handling higher forces, therefore I use them very often. Especially  "T5" is more or less my standard - the T profile with 5mm pitch.

I placed a feature request in "freecadweb.org", too, but I was suggested to do it also here.

**Why T profile:** 
The T-Profile is a very simple one, but it is also very common in Europe. It has a little bit less precision (compared to GT), but it helps nothing to be able to print GT gears and have almost no chance to buy a belt for it. So at least T5 would help me a lot, but there are also smaller pitches (T2, T2.5, ...)

My current way therefore is to use openscad (there a T5 generator exists), create only a ring of teeth with a big hole in center, export that as .obj, import that in Freecad, and embed it into my part (create a volume part from .obj).   This is extremely complicated and often leads into FreeCAD crashing when I need to modify this part made of thousands of meshes. 
_This is no real fun._

It would be very helpfil if I could create a T5 timing gear directly in FreeCAD.

For your reference, I attach an openscad drawing. This is a timing gear with 110 teeth. As it is configured, it perfectly fits T5 belts.  If you need help, I can offer you some testing. I can print timing gears, test them with existing timing belts of T5, and give you feedback.  Sadly it exceeds my knowledge on how to write code, so I cannot help you with extending your great FCGear module.

[Pulley_Generator-110Z-T5.scad.zip](https://github.com/looooo/freecad.gears/files/6106837/Pulley_Generator-110Z-T5.scad.zip)
Here I found a technical drawing for T5 profile
[https://www.google.com/url?sa=i&url=http%3A%2F%2Fwww.nk-gmbh.de%2Fprodukte%2Fzahnriemen%2Fmetrische-teilung%2Ft5-profil%2F&psig=AOvVaw39RsTE4KAhQ_h6Z6Kw4w2h&ust=1615363724713000&source=images&cd=vfe&ved=0CAIQjRxqFwoTCMCS5fngou8CFQAAAAAdAAAAABAE]

azalanono, 2021-03-09:
I made a test right now - to show you how crazy it currently is:

1. create the timing gear profile in openscad
2. export as .stl file
3. import the .stl in FreeCAD 
4. convert mesh into shape
5. convert shape into volume body

And now you have "something" that looks like a timing gear, but is a nightmare in detail. You can see this by all this lines on surface going in all directions.  You are absolutely unable to chamfer teeth, of drill a hole between teet (for worm screw). If you try, it is very likely that FreeCAD crashes hereby or does something wrong.

Therefore it is very important to me to get a plain, smooth part directly, all other things are - friendly named - crazy workarounds.

AKielhorn, 2021-04-01:
I would like to second that request.

There is a whole family of timing belts:  T2,5, T5, T10 and T20.

There is a similar family AT3, AT5, AT10 and AT20 for higher transmission power.

These two families are metric, the corresponding imperial sizes are XL, L, H and XH.

And in order to make it more interesting there are T1/5", T3/8", T1/2" and T7/8" as well.

You can find drawings of the profiles at [Brecoflex](https://www.breco.de/en/downloads).

I have a parametric model that I use as a template right now, but having these in the gears workbench would be a great improvement.

AKielhorn, 2021-04-01:
I suggest a separate class ` TimingGearT` for trapezoid.

Will look into it over the Easter Break but can't promise nothing.

herrgerd, 2021-05-11:
> 
> 
> I made a test right now - to show you how crazy it currently is:
> 
>     1. create the timing gear profile in openscad
> 
>     2. export as .stl file
> 
>     3. import the .stl in FreeCAD
> 
>     4. convert mesh into shape
> 
>     5. convert shape into volume body
> 
> 
> And now you have "something" that looks like a timing gear, but is a nightmare in detail. You can see this by all this lines on surface going in all directions. You are absolutely unable to chamfer teeth, or even drill a hole between teeth (for worm screw), not to mention subtracting more complex parts from this "something", i.e. a reciption for a shaft.
> If you try, it is very likely that FreeCAD crashes hereby or does something wrong.
> 
> Therefore it is very important to me to get a plain, smooth part directly, all other things are - friendly named - crazy workarounds.

You can generate the openSCAD model directly in FreeCAD so you don't have to go the "import .stl" route. See [here.](https://wiki.freecadweb.org/Import_OpenSCAD_code/de)

azalanono, 2021-05-12:
Hello, herrgerd.
Thank you for this suggestion. 

**There is only one problem:**  
I have tried to do that more than one time. It **never** worked.
This may work on Windows with fully installed programs, but I am running Linux (openSUSE Leap 15.2 x64) and have only appimages. Each appimage for itself works (openscad and freecad), but I never managed to import an openscad file into freecad.

Even when this would work (I would be happy with that), it is still a workaround in my eyes. Sadly it does not work at all, therefore my request.

herrgerd, 2021-05-12:
> Hello, herrgerd.
> Thank you for this suggestion.
>
> **There is only one problem:**
> I have tried to do that more than one time. It **never** worked.
> This may work on Windows with fully installed [...]

There was a period last time I checked where there was a missing dependency called ply: https://wiki.freecadweb.org/OpenSCAD_Module

Seems fixed in windows, maybe installing it does work for you?

azalanono, 2023-12-31:
Hello, looooo.

I noticed right now that I posted this two years ago. Yesterday I had to re-install the FCGear workbench because I upgraded to FreeCAD 0.21.1, so I also checked if (and what) has changed in FCGear. It appears to me that absolutely nothing has changed, I still cannot find any profiles for trapezoid timing belts.

Would be great if trapezoid profiles will find into your AddOn, because all other things it offers are really useful, only this different profiles are missing.

looooo, 2023-12-31:
>  So it would be better to do that during generation of the gear.

I am not sure how to do this. And I think there is not much improvement if this is done via scripting. Best is to create a cylinder with same size as the gear an chamfer the edges. Then make a boolean intersection to get the chamfered gear.

you can access the properties via:
InvoluteGear.da / 2 for the radius of the cylinder
InvoluteGear.height for the height of the cylinder 

I tried this with a gear with 200 teeth and it was computed in 20
[image]
 sec.

looooo, 2023-12-31:
@azalanono do you have a drawing of t5 gear or belt? What properties are needed? For the other timing gears we have only height and number of teeth. Is that enough?

azalanono, 2024-01-01:
Happy New Year to everybody!

Looooo, What you show in your screenshot is my current approach. Sadly this bears the problem that teeth can still be overextruded in the lower region, where they touch the build plate. The same happens much more drastically when printing the gear on resin printers (-> elephant leg on first layers because of over-exposure). 
This burr prevents correct matching of two gears. As solution I have to manually file away this burr which is no fun at 100-200 of small teeth.
What you describe/suggest I currently do on entire tooth therefore, but this means that the tooth is printed without support, because support is not possible. to not loose too much width, I have to use very little angle, this results in poor appearance.

Therefore my idea is to chamfer the entire tooth along its complete shape only a very little bit (i.e. 0.4 - 0.5 mm).  So the core of tooth is starting on build plate, but because if its recess it is printed without a burr. See below picture. this is how I imagine the final tooth.

[image]

Of course, this chamfer has to be on both sides of teeth.   

Trying that wit gears of >100 teeth and undercut etc. almost always makes FreeCAD crash when I try that by chamfering an area (so all edges are marked automatically).  So it is for sure smarter to do this chamfering during generationg the teeth.

Advantage of this way: ll teethstart flat on building plate, but the chamfer (over-)compensates otherwise occuring overextrusion/elephant leg completely, I do not have to use a fine file to clean all teeth by hands after printing.

Hope I could explain what I mean.

looooo, 2024-01-01:
I am sorry but this is not possible with how the gear is created. The creation of the gears should be fast, and such an operation is time-intensive with both python-scripting or freecad-gui.

azalanono, 2024-01-01:
Hi, looooo.

**It is nice when a gear is created fast. But when the result is more or less useless afterwards, that makes no sense at all.** 

So at the moment I can choose my poison: 
I can generate gears very fast. Afterwards I have to make them **somehow** usable by building the "common" of generated gear and a cylinder chamfered on both ends. Chamfering has to be in full tooth height to be sure that no burr is remaining, but bears the problem of printing almost horizontal surfaces without support...  In addition, this are then two steps. As FreeCAD is not able to insert some modification afterwards (somewhere in the tree), this is not really practical at all ...

... or ...

I try (and pray that FreeCAD doesn't decide to do crazy things or to crash when I try) to apply chamfers afterwards in FreeCAD. Hereby it is highly recommended to save the design first, because everything else is really risky. Again, as FreeCAD is not able to insert some modification afterwards (somewhere in the tree), this is not really practical at all ...

I think both options are not what a user really desires, and having to wait additional time for a perfect result will be preferable in many cases. Also, one can design all parts first, and activate chamfering in a final step.

**You could add this additional chamfering in a way, allowing to turn it on when needed.** So your gear generation will have same speed by default, but I can optionally set chamfer width to a value >0.00 (hereby turning chamfering on), and wait a bit longer for a final perfect result.

Wouldn't that be the perfect solution?

The major problem I watch is:
When enabling undercut and possibly slightly increasing the quality in addition, the program generates an object that is sometimes too complex for FreeCAD. I couldn't find out where the critical point is, but it repeatedly happens.  FreeCAD often crashes when I try to add the chamfer afterwards.

looooo, 2024-01-01:
Isn't there an option to remove this "elephant feet" with the slicer?
https://help.prusa3d.com/article/elephant-foot-compensation_114487

azalanono, 2024-01-01:
Looooo,

using resin printers causes LOTS bigger elephant foots than using FDM 
printers. This is caused by over-exposing first layer to increase sticking to build 
plate. Especially when having wider solid areas, this is extreme.

The reason for using resin printeres is: I can make smaller gears because I 
have higher resolution and can make finer details.

Chitubox slicer allows compensation by reducing size of base layers, but this is 
not really perfect. It improves the result a lot, but not everything.

azalanono, 2024-01-01:
But lots more important is generating pulleys for timing belts like T2, T2.5, T5, ... (Metric, trapezoid teeth), so this are more or less missing profiles...

I wrote some links at the very beginning of this thread, pointing to dimensions, too.

looooo, 2024-01-01:
[image]

one fast workaround is to create 4 gears with height set to zero (2d wire). Then position them and set the module to a smaller value for the outer two wires. Then use part loft (solid, ruled). Works pretty fast for a test case with 100 teeth. With spreadsheet or dynamic data it's easy to make this loft parametric, computing the position of the gears, the module and backlash of the reduced gears.
I hope this resolves this issue.

looooo, 2024-01-01:
> But lots more important is generating pulleys for timing belts like T2, T2.5, T5, ... (Metric, trapezoid teeth), so this are more or less missing profiles...
> 
> I wrote some links at the very beginning of this thread, pointing to dimensions, too.

the link doesn't work anymore. And I can't open the opens cad file

azalanono, 2024-01-01:
I attach the .openSCAD file here. It creates a perfectly matching T5 
timing belt gear for a sewig machine. 

It is plain ASCII, should be readable therefore. In worst case 
rename it to ".txt" if you work on Windows.

Perhaps it is encoded in UNIX style, then <CR> or <LF> is missing.

azalanono, 2024-01-01:
Here now the .scad file.  The email attachment wasn't added before.

[Pulley_Generator-88Z-T5.zip](https://github.com/looooo/freecad.gears/files/13805193/Pulley_Generator-88Z-T5.zip)

looooo, 2024-01-01:
sorry I meant, I have never used openscad and therefor don't know what to do with this file. A drawing with the dimensions would be more helpful.

azalanono, 2024-01-01:
Ok, I found this drawing for T5 timing belt:

I try by embedding in mail body and also as attachment...
[1]

From my experience with openscad...  It is helpful if the ratio between timing belt tooth-
gap and timing belt tooth-width can be adjusted.  
Pulleys usually have little narrower teeth compared to the timing belt tooth-gap, so the 
timing belt has a very little free play in its moving direction and finds easier into the pulley.

This is also necessary when using a FDM printer because of its nozzle diameter.

If you want me to test one or other generated pulley, please send me the stl of pulley.  I 
can print it then on both:  resin printer and fdm printer and give you some results.

--------
[1] Bild

azalanono, 2024-01-01:
Here the link to the excellent web site (and datasheet with lots of technical 
details:

https://www.nk-gmbh.de/wp-content/uploads/2017/07/Produktkatalog-Zahnriehmen.pdf[1]

It would be great if you could implement T2.5 (2.5mm pitch),  T5 (5mm pitch) 
and T10 (10mm pitch).

--------
[1] https://www.nk-gmbh.de/wp-content/uploads/2017/07/Produktkatalog-Zahnriehmen.pdf

looooo, 2024-01-01:
@azalanono I added a new command for the desired timing gear. It is really raw, but maybe you can test it if it fits your needs.

azalanono, 2024-01-02:
Hi, looooo.

Thank you a lot that you jumped in so quickly with an update.
Sadly my knowledge on how to install from refactoring branch is zero.  Can you, please, tell me how to install this version (in case that only there the T5 timing gear is added) ?

I use the Linux AppImage of FreeCAD 0.21.1  on Linux.

looooo, 2024-01-02:
I have merged the refactoring diff to master in the hope I didn't break anything. But I wanted to do this anyways...

So simple try the master-branch.

azalanono, 2024-01-02:
Hi, looooo.

I have done a very first try with 15 teeth on my  Anycubic Photon S  resin printer. (a simple T5, 15 teeth, 3mm height)  and want to give you some feedback.

- In general, the pulley matches the belt very well. Especially with small diameters it often happens that the belt's teeth collide a bit at joining and separating points.   So what you did is really good.

What I can see having a closer look:

- There is a little gap between tip of pulley-tooth and timing belt (between teeth). This is not so good, because this means, the belt is running on a slightly bigger diameter. This causes issues on bigger gears.. The belt should sit with the flat section between its teeth on top of pulley teeth.  
- In FreeCAD I can see that the pulley has sharp edges at tooth-tip and tooth-bottom. if you apply a slightly bigger radius an tips as the belt needs, there will be no collision any more (which possibly causes this gap).
- I would also apply a little radius at bottom of pulley-teeth. A little less than radius of timing belt tips. This rounding makes the pulley-tooth a bit stronger and also makes printing on FDM printers faster, as no full stop is forced by a sharp edge.

- What I am scared about is printing pulleys on FDM printers.   Usually this printers add some material when they are moving slower this is normally the reason for drawing holes i~50% of nozzle-diameter bigger to make them fit.  If the same happens with the gap between pulley-teeth, this will become a problem because the belt will not fit between teeth!
I experienced that belts often perfectly fit on small pulleys, but not on huge pulleys, when teeth do not fully fit into the pulley. The fault caused by bigger radius accumulates and the belt will not join and separate correctly. (I had this often with timing gears of 200+ mm diameter).

So in brief:
For first try - on my resin printer - a surprisingly good result, almost perfect. I haven't tested on FDM printer yet.

If you could add a mechanism for adjusting the with of pulley-teeth, the result will be absolutely perfect. I imagine two values either as multipliers of offsets:   "Tooth-width correction" and "Tooth-height correction". If this are offsets or multipliers is not so important, but multipliers appear to be easier.   
I.e.   Tooth-width correction:  0.8    and   Tooth-height correction: 1.2

looooo, 2024-01-02:
Thanks for the feedback. I wanted to add the fillets anyways, but its good to inow that the main geometry works.
I thought about implementing the teeth correction by a "backlash" parameter which is also available in other gears. What do you think?
The height of the tooth is already parametric and can be adjusted.
But maybe we take a similar approach like for the other timing gears (parameters given by the type of gear) and add a vorrection factor like you said.

I have one more question:
Should the "head" and "foot" of the gear be modeled as arcs or straight lines?

azalanono, 2024-01-03:
Hi, looooo

Whatever you choose, it should allow adjusting height and width of teeth intuitively (so 
easy understandabla naming is very welcome.

About last question:  
Arcs are always better than straight lines. Keep in mind that it should be possible to finally 
chamfer the pulleys along its teeth.  Always keep surfaces in a shape not needing further 
editing, because FreeCAD is very limited.

Now they brought out 0.21.1 and it is still not possible to edit something in an object 
without keeping  the previously set transparency and color as is. The settings are there, 
but not applied again. You have to modify them back and forth to re-apply them... This sais 
enough...  Not to talk about chamfering and/or rounding edges. If you afterwads change 
some dimension(s) and hereby the number of edges changes, it is completely lost or 
applies rouchings/chamfers to wrong edges.

looooo, 2024-01-04:
addes:
- backlash (makes the teeth smaller by the length in mm on the pitch circle)
- root_fillet: add fillet on the tooth-root (this is not working, somehow it chooses the wrong midpoint)
- head_fillet: add fillet for the tooth-head

also I converted the straight lines to arcs.
So only missing thing is the reversed root_fillet. But I have no idea how to solve this. I think this algorithm is a numerical minimization which starts at a given point p0 and the looks for a point that is normal to both curves and has the same distance. But somehow it finds the wrong midpoint...

looooo, 2024-01-04:
Further you are maybe able to find good parameter for different pulleys. If you found them we can similarly to the gt profiles add a dict with default parameters for different t-pulleys.
And we also need a good icon for this new gear type. maybe you have an idea.

azalanono, 2024-01-04:
Am Donnerstag, 4. Januar 2024, 14:36:06 CET schrieb lorenz:
> Further you are maybe able to find good parameter for different pulleys. If
> you found them we can similarly to the gt profiles add a dict with default 
> parameters for different t-pulleys. 

This is a bit tricky, because a well fitting pulley needs some backlash, but not too much.  But 
there is a massive difference between FDM printers and resin printers.  
A FDM printer tends to do some over-extrusion when it suddenly has to reduce speed, and 
tends to under-extrusion when starting a path.
A resin printer simply stacks pictures on top of each other and doesn't have such issues 
therefore.  
But resin tends to expang on long exposure times massively (reason for elephant legs 
when printing directly on build platform, as there the exposure time is 10 times longer).

I would go a different way, if this is easily possible:
Allow to save the settings back to the profile and add there a "Return to defaults" to return 
to profile defaults.  Hope I explained clear enough.  No hardcoded-only profiles, but 
allowing to overwrite them (defaults still hardcoded in background).

> And we also need a good icon for this new gear type. maybe you have an idea.

This is easy:
All yout GT pulleys have half-round belt-teeth, while T pulleys have trapezoid belt teeth.
Also the name is a bit different:  GT vs. T.

I would subdivide the icons horizontally into 2/3 + 1/3. Then I would write "GT"  or  "T" into 
upper section.
In lower 1/3 rd I would place a white picture of tooth profile. Imagine a white straight timing 
belt with teeth facing downwards. Now magnify one tooth . What you get is a white stripe 
from left to right, with either a half round or trapezoid tooth faciong downwards below.

That's it!

-- 
 
mfg
 
  Ing. Rainer Hantsch
    (Firmeninhaber)
 
-------------------------------------------------
 ING. RAINER HANTSCH - Hardware & Software
 Khunngasse 21/20
 A-1030 Wien
 *** Austria ***
 
 Tel.: +43-1-7988538-0     Fax : +43-1-7988538-18
 Mobil: +43-664-9194382    Signal-Messenger (kein WhatsApp)