looooo/freecad.gears #67: Cycloidal disk should be contracted.

iplayfast, 2021-05-02:
I really like what you've done with the cycloidal disk. I believe you are using a normal cyloidal instead of contracted (which is actually the useful one, less vibration).
This website gives a good overview + math formulas to implement.
https://www.tec-science.com/mechanical-power-transmission/planetary-gear/construction-of-the-cycloidal-disc/
For reference this one is also good.
https://stepbystep-robotics.com/hp/robots/cycloidal-drive/

lmunier, 2021-07-14:
Hello iplayfast,

Since I am also interested to have a contracted cycloidal gear, I compared the actual version of the workbench implementation with the one from step-by-step robotics (and also with the one from this paper : https://asmedigitalcollection.asme.org/mechanicaldesign/article-abstract/132/2/024503/431798/Unified-Approach-to-Cycloid-Drive-Profile-Stress?redirectedFrom=fulltext). And everything seems to give the same results except that the workbench implementation output a rotated gear, but with the same profile as the two others.

I am assuming that a contracted version could be obtained using the right parameters (seems that the eccentricity influences it a lot).

Have you others suggestions about that?

Best regards,
lmunier

iplayfast, 2021-12-02:
Sorry I didn't get back to you on this, must have missed the email. 
I'm having problems loading workbenches currently but if I recall, 
https://www.tec-science.com/wp-content/uploads/2021/03/en-cycloidal-drive-speed-reducer-construction-comparison-ordinary-contracted-cycloid.jpg
Shows the difference between the two types. My project https://github.com/iplayfast/CycloidGearBox creates a gear box, but the main math is in cycloidFun.py I think it provides what you need.
It's been a while since I visited this problem, so it's quite possible that I'm totally mistaken in this bug report. 
I did notice however that it's hard to actually use the cycloidal gear in actual projects. This might be due to my lack of skill in FreeCAD, but, see if you can make a gear that fits inside a 100MM square. and then pad it etc to use inside a gearbox. 
Sorry I couldn't be more helpful.