looooo/freecad.gears #71: Bevel gear issues

dperry20, 2021-06-22:
Tried working with bevel gear. Many issues. First if you change the height of the bevel gear, the angle which is suppose to remain at  
45 degrees also changes.
Second if you change the Beta angle and the height, you end up with two beveled gears. 
Third the base of the beveled gear is flat. This is not common to other beveled gears I printed from other sources.

Great work on the gear workbench.

looooo, 2021-06-23:
The gears are meant to be basic geometries to be combined with other geometry by boolean operation. There are some limitations to the parameters. It's important to understand the correlation of a parameter on the model and to know the limitations otherwise strange results can be the output.
So simple create the geometry you want with basic freecad elements (sketcher, Partdesign, Part) and use the bevel gear to cut the teeth from the geometry. That's how this workbench is supposed to work.

dperry20, 2021-06-23:
Thank you. I realized this after watching some videos. There are some bugs though. If you make the height to big, the angle does not remain the same. I am wondering to what extent can you change the height without affecting the angle. 
Thanks for responding.

looooo, 2021-06-24:
here is an example how to calculate the max_height. maybe we can add a warning if the height parameter is too big.

[image]

looooo, 2021-06-24:
The conditional warning was added here: https://github.com/looooo/freecad.gears/commit/dff8ee7b424362c89cb5e9fe2c828b2b014469dd#diff-637bc0e7541865d9034069165058469578ee67325d5e16ec5e276f131b24ebd4R611

luzpaz, 2023-02-05:
@looooo please close ticket.

closed 2023-02-05 by dperry20