looooo/freecad.gears #76: problem adding holes to a gear via a sketcher

brown-richard, 2021-07-10:
This is a great tool for making gears. I seem to have a problem adding holes to a gear. I get errors when I try to do this:  
1) go to the sketcher workbench 
2) highlight the face of the gear
3) click on add sketch. 
The errors are: 
17:00:08  AttachEngine3D::calculateAttachedPlacement:InertialCS: inertia tensor has axis of symmetry. Second and third axes of inertia are undefined.
17:00:08  Sketcher::SketchObject / Sketch: Links go out of the allowed scope
17:00:08  AttachEngine3D::calculateAttachedPlacement:InertialCS: inertia tensor has axis of symmetry. Second and third axes of inertia are undefined.

I tried just creating the gear and adding the holes and I received errors doing that. If you could look at the tutorial Awesome Gears in FreeCAD by a person out of the UK maybe that would help. Not much more I can say. I am new to FreeCAD and looking at all the tutorials I can find. Great CAD program, maybe I am doing something wrong. Thanks for your understanding. Richard Brown

looooo, 2021-07-17:
Can you provide more information (freecad version). I cannot reproduce with:

```
OS: Ubuntu 18.04.5 LTS (ubuntu:GNOME/ubuntu)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.20.24562 (Git)
Build type: Release
Branch: master
Hash: 8277b3f9e9b66092a58625f3a2a87a0d60dcc580
Python version: 3.8.6
Qt version: 5.12.5
Coin version: 4.0.0
OCC version: 7.4.0
Locale: German/Germany (de_DE)

```

brown-richard, 2021-07-17:
Thank you for replying to me. I have tried this tutorial many times, at least 7 or 8 times. I have attached two version of the attempts. This is the tutorial I am trying  https://www.youtube.com/watch?v=Wu-zGDy7WQI&t=526s

The person who made the tutorial is of NO help. I have python on my machine and looked at it a year or so ago. I do not think this could be a problem, but maybe it is. If there is anything else I can do to help resolve this, please let me know. I must be doing something wrong. 

Again thanks for your help

Richard 

OS: Windows 7 Version 6.1 (Build 7601: SP 1)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.19.24291 (Git)
Build type: Release
Branch: releases/FreeCAD-0-19
Hash: 7b5e18a0759de778b74d3a5c17eba9cb815035ac
Python version: 3.8.6+
Qt version: 5.15.2
Coin version: 4.0.1
OCC version: 7.5.0
Locale: English/United States (en_US)

On Saturday, July 17, 2021, 03:51:43 AM PDT, lorenz <notifications@github.com> wrote:

Can you provide more information (freecad version). I cannot reproduce with:
OS: Ubuntu 18.04.5 LTS (ubuntu:GNOME/ubuntu)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.20.24562 (Git)
Build type: Release
Branch: master
Hash: 8277b3f9e9b66092a58625f3a2a87a0d60dcc580
Python version: 3.8.6
Qt version: 5.12.5
Coin version: 4.0.0
OCC version: 7.4.0
Locale: German/Germany (de_DE)

[[image]
[image]

brown-richard, 2021-07-17:
I will try to upload the FreeCad files by renaming
[gear_nbr_03A.FCStd.txt](https://github.com/looooo/freecad.gears/files/6835391/gear_nbr_03A.FCStd.txt)
[Gears_New_01C.FCStd.txt](https://github.com/looooo/freecad.gears/files/6835392/Gears_New_01C.FCStd.txt)
 the extension ..

brown-richard, 2021-07-17:
Thanks again. Richard Brown

closed 2021-07-17 by brown-richard