looooo/freecad.gears #78: Can this make a planetary ring gear?

daveseff, 2021-08-14:
Trying to design a planetary gear assembly. I can not see a way to create the outer ring gear. Can this module create one?

pitch-circle, 2021-09-29:
The outer ring gear of a planetary gear is known as an internal gear. Easy reference on it is here: 

http://vintagemachinery.org/pubs/detail.aspx?id=5659

Comprehensive reference in here:

http://qtcgears.com/tools/catalogs/PDF_Q420/Tech.pdf

To make an internal gear, make a standard spur gear, superimpose a cylinder over it and do a Boolean subtract.

It is the same thing as making a centre hole in an external gear - in that case you subtract the hole from the gear.

looooo, 2021-09-29:
In addition @jbaehr is working on this. please have a look at https://github.com/looooo/freecad.gears/pull/77

daveseff, 2021-09-29:
Thanks. In the end I used a spur gear with a boolean to make the outer "internal" gear. Thanks.

closed 2021-09-29 by daveseff