looooo/freecad.gears #79: Asymmetric Involute Gear Profile?

charmquark1984, 2021-08-14:
I have found the FCGear workbench extremely powerful and useful for my robotics projects so far. 

I wish to design gears with different pressure angles on the leading and trailing flanks. In my case I want to print a vertical rack without exceeding the maximum overhang angle of my 3D printer (and the highest practical symmetric pressure angle of 30 degrees is insufficient), but people also do this in industry to improve the strength of gear teeth for gears loaded in only one direction. 

Would it be possible to modify the existing involute profile generator to support asymmetric teeth? 

[https://gearsolutions.com/features/designing-asymmetric-gears/](https://gearsolutions.com/features/designing-asymmetric-gears/)

pitch-circle, 2021-09-29:
Apart from yourself, how many other users of this module would use this feature?

Can you not create two gears, each with the correct geometry, delete 50% of each and join them together to achieve the result you desire?

looooo, 2021-09-29:
you can do this with a boolean common (Part-workbench). Create one gear with 20° pressure angle and another with 30° pressure angle and rotate the second  gear to align with the first. Then use boolean common to get the desired result. To make the teeth thicker you can also use the backlash property.

[image]