looooo/freecad.gears #81: Part Design integration

mwganson, 2021-10-02:
Some time ago I remember having a discussion with you on putting solids into a Part Design body and have it all work correctly.  I didn't know the answer at the time, but I have learned a few things since then.  See my PDWrapper macro (and also Bevel macro) for some code examples.  I haven't put PDWrapper on github yet.  Bevel is in the addon manager.  I have topics for these currently on the scripting forum.  PDWrapper is a wrapper object for non-PD objects so they can be used in PD.  I haven't tested your gears, but they would probably work with PDWrapper.  Bevel is a dressup similar to chamfer except it works on vertices instead of edges.

Your gears would be more or less the same as the additive/subtractive primitives available in PD.  Using obj.addExtension("Part::AttachExtensionPython") would make them attachable if they are not already.

In a nutshell, how to integrate into PD:

* base object on PartDesign::FeaturePython (or PartDesign::FeatureAdditivePython / PartDesignFeatureSubtractivePython to be able to use them as base for pattern features, such as in a polar array)
* use body.newObject() instead of doc.addObject()
* for additive/subtractive features add a Part::PropertyPartShape property named "AddSubShape", the shape to be used for copying in a pattern
* the object's Shape property must be fused to (or cut from) the previous solid feature (body.Tip before adding your feature)
* AddSubShape will be the shape you would normally have in a Part::FeaturePython object
* Sometimes body.Tip will be empty.  You can parse the features in body.Group and test each feature's volume against the volume of Body.Shape to see which is the current tip.  I haven't used this, but it should work the vast majority of the time.  You could also probably just use the last feature in the body.Group list, just make sure it's not a shapebinder, sketch, etc.  If your object is the first feature, then AddSubShape and Shape will be the same.

looooo, 2021-10-02:
@mwganson Did you already test the PartDesign integration from @jbaehr ?
https://github.com/looooo/freecad.gears/pull/72

mwganson, 2021-10-02:
I did, but only minimally.  I did another test just now.  What you have, I think, is working very well.  I am impressed with the attachment functionality.  I see now you have already done bullet point #4, fusing to the previous solid.  

Still not working: 1) pattern support, 2) subtractive option.

For pattern support (for example 3 planetary gears in a polar pattern around a sun gear) you would have to base the gear on PartDesign::FeatureAdditivePython and implement the AddSubShape property (which is only the gear shape itself not fused with the previous solid feature).

I think a subtractive option (without pattern support) would be fairly easy to implement and you can continue using Part::FeaturePython.  I suggest adding a property of type App::PropertyEnumeration = ["Additive","Subtractive"] set to the default value of "Additive":

```
obj.addProperty("App::PropertyEnumeration","Type","GroupName","ToolTip").Type = ["Additive","Subtractive"]
obj.Type = "Additive" #default

```
Then in execute() when creating the shape if type = Subtractive you would cut the gear shape from the previous solid feature rather than fusing to it.  The user can easily toggle from Additive to Subtractive should he wish to do so.  How useful this would be in actual practice, I do not know, but it's fairly easy option to add.

jbaehr, 2021-10-02:
@mwganson There is a follow-up to the already mentioned #72 (make gears attachable): #74 (improve PartDesign support). When starting the work on the later, I also thought that providing "subtractive gears"  is easy. And it is from a developer's perspective. For the user though, such a straight-forward implementation would result in a complete mess. Please see my musings on this topic in the description of #74, the last, unchecked bullet. I actively decided against doing it this way. (I'm not against "subtractive gears" in PD::Bodies per se, quite the contrary. I do want them, but I want to do them right. So that they really serve the user, following the same spirit as the other PD tools. Not only as a programming exercise "because we can".)

Regarding pattern support: Cool, I did not now about "PartDesign::FeatureAdditivePython" / "PartDesignFeatureSubtractivePython". I fail to see the use case for gears, though. You mention the planets in an epicyclic gear train. But you would never model a complete gear train as a single PD::Body (which must result in a single solid!). Rather, you'd model each single gear type (sun, planet, annulus) as individual bodies and later assemble them somehow.
On a second though: When making a casting mold for gears, patterns (and subtractive gears) could become handy, though.

mwganson, 2021-10-03:
Excellent.  I only recently discovered these things myself and wanted to share them with Looooo as he has helped me in the past.  Perhaps they will be useful in some future project if not in this one.

mwganson, 2021-10-03:
I need to correct myself on something.  I have recently learned that there is no need to create a AddSubShape property with the additive/subtractive types.  They will already have this property.  Adding a new one just creates AddSubShape1 unnecessarily.

jbaehr, 2021-10-03:
@mwganson Would you mind creating an article in the FreeCAD wiki on how to extend PartDesign objects using Python? I remember having a hard time finding all the relevant stuff needed for #74 (links to my sources in the respective commit messages). And still, this is the first time I hear about "PartDesign::FeatureAdditivePython", which seems to be the better choice here...

mwganson, 2021-10-04:
I am planning to do this.

MariwanJ, 2021-11-12:
@ mwganson
Please would you mention that the object you created is not a FreeCAD object. It is COIN3D object which has no shape and cannot be manipulated by FreeCAD tools. Otherwise it could mislead other new programmers. 
thanks

mwganson, 2021-11-13:
I don't understand.   In what manner would you like to manipulate the tube with FreeCAD tools that you cannot do?

mwganson, 2021-11-13:
I think I see now.  I only added the Part Design section at the bottom of that page, with the code for creating the tube primitive.  The COIN stuff was already there.