looooo/freecad.gears #86: Make clearance apply to teeth shape?

scottmudge, 2021-11-21:
Is it possible to make the clearance parameter apply to the outer surface of the teeth? With the cycloid gear type (at least, maybe others), the clearance parameter only seems to affect the inner core diameter of the gear, but not the width of the teeth or the distance the teeth protrude. 

I've tried performing the "3D offset" operation to achieve this effect, but it is quite slow and does not always work.

Is it possible to add a "global" clearance? It's possible to achieve a similar affect by changing the module slightly, but I would like to keep my module values at their ideal values.

looooo, 2021-11-22:
Have you tried the `backlash` parameter?

scottmudge, 2021-11-22:
Yes but it did not have the desired effect either. I'm more looking for a global clearance/offset, to make the generated object/model more suitable for FDM. 

I can create an interface in a spreadsheet to convert the module value to one which conveys a global clearance:

`New Module = (Diameter - Clearance) / Teeth Count`

This seems to work fine, but it's annoying to manually create a parametric converter/interface. I may make a fork which performs this operation automatically on a new "global clearance" parameter.

scottmudge, 2021-11-23:
I'll close this for now and create a PR if I find a suitable solution that makes sense in context with the current implementation.

closed 2021-11-23 by scottmudge