looooo/freecad.gears #87: Development branch -- Saving and reloading project yields "object has no attribute `icon_fn`

scottmudge, 2021-11-22:
I'm using the development branch for the internal gear feature, but notice that when I try to reload a saved project, the icons in the tree view do not appear correctly, and the following error shows up in the logs:

```
16:51:51  Traceback (most recent call last):
  File "C:\Users\Scott\AppData\Roaming\FreeCAD\Mod\fcgear\freecad\gears\features.py", line 66, in getIcon
    return self.icon_fn
<class 'AttributeError'>: 'ViewProviderGear' object has no attribute 'icon_fn'
16:51:52  Traceback (most recent call last):
  File "C:\Users\Scott\AppData\Roaming\FreeCAD\Mod\fcgear\freecad\gears\features.py", line 66, in getIcon
    return self.icon_fn
<class 'AttributeError'>: 'ViewProviderGear' object has no attribute 'icon_fn'
```

Indeed, the code in question was refactored between master and development branches. I thought it might be due to the workbench not being loaded when the project was opened, but even loading the workbench before opening the project yields the same result.

looooo, 2021-11-23:
can you try my latest commit? https://github.com/looooo/freecad.gears/commit/15f3f2410dead46105a1d3acd8588a8d8f496f0e#diff-637bc0e7541865d9034069165058469578ee67325d5e16ec5e276f131b24ebd4L69

Should fix this issue.

scottmudge, 2021-11-23:
Yes it corrected the issue for new projects, but I was unable to successfully reload my old projects which had this issue. When attempting to do so, the following two errors occur for each gear in the scene:

```
14:43:03  Traceback (most recent call last):
  File "C:\Users\Scott\AppData\Roaming\FreeCAD\Mod\fcgear\freecad\gears\features.py", line 72, in __setstate__
    self.icon_fn = state["icon_fn"]
<class 'TypeError'>: 'NoneType' object is not subscriptable

...

14:43:03  Traceback (most recent call last):
  File "C:\Users\Scott\AppData\Roaming\FreeCAD\Mod\fcgear\freecad\gears\features.py", line 66, in getIcon
    return self.icon_fn
<class 'AttributeError'>: 'ViewProviderGear' object has no attribute 'icon_fn'
```

The only way to get rid of them is to delete the gear and recreate it manually as a new gear.

It's probably too much work to add a compatibility layer to compensate for this, but it's something to consider for forward compatibility.

scottmudge, 2021-11-23:
Alright I implemented a fix for broken projects, creating a PR now.

scottmudge, 2021-11-23:
PR ready, closing for now unless issue reoccurs.

closed 2021-11-23 by scottmudge