looooo/freecad.gears #98: Helical internal involute gear

ofpkubi, 2022-01-14:
Hi,

It seems when I give a beta other than zero (even if 0.001 degree) for an internal involute gear, it loses its proper tooth geometry and goes into a plane walled tooth.

**Reproduce**
Please see the attached file. Just change the beta in the spreadsheet.
[planetary.zip](https://github.com/looooo/freecad.gears/files/7873681/planetary.zip)

**SW version(s) used**
Debian testing
Stable: FreeCAD_0.19.3-Linux-Conda_glibc2.12-x86_64
Weekly: FreeCAD_weekly-builds-27008-Linux-Conda_glibc2.12-x86_64
Realthunder's: FreeCAD-asm3-Daily-Conda-Py3-Qt5-20220111-glibc2.12-x86_64

Development branch of freecad.gears. Installed on 2022-01-14 from github (pip install https://github.com/looooo/FCGear/archive/develop.tar.gz)

looooo, 2022-01-15:
@ofpkubi can you post a screenshot?

ofpkubi, 2022-01-15:
Here you are @looooo .

15, 44 and 60 teeth
left side: beta=0 degree
right side: beta=1degree

[image]

looooo, 2022-01-15:
I guess this is related to the deviation parameter. Try to set it to a smaller value.

ofpkubi, 2022-01-15:
Which one is the deviation parameter?

looooo, 2022-01-16:
in the view tab, there is a deviation property, see the image.
you can also set it globally in setting, part-design, second tab

hope this helps.
[image]

ofpkubi, 2022-01-16:
Thanks!
Now, this is interesting.
Originally the deviation is 0.2 on all objects. If I change it to 0.05 on the large gear, it shows a better tooth profile. But, if I change it back to 0.2, the toot profile stays good.
But anyway, this seems to be only a viewing issue, the exported STL looks good.

I think there is no bug then.

looooo, 2022-01-21:
@ofpkubi if it is solved, please close this issue. thanks

ofpkubi, 2022-01-21:
Done. Thanks a lot!

closed 2022-01-21 by ofpkubi