looooo/freecad.gears #100: Bevel Gears - Incomplete tooth forms @ Pitch Angles > 50°

rdtravis, 2022-02-18:
I'm a newbie in FreeCad, and using your gear module to model a 13tooth pinion/44 tooth ring spiral bevel gear pair, which I'm then exporting into solidworks.  It seems that for pitch angles greater than 50°, the outer edges of the lower tooth flanks are running off the part, resulting in something resembling a saw cutter.

Pitch Angle = 73.54° [image]

Pitch Angle = 30° [image]

I was able to extrapolate/project the missing sections, and while the gear looks somewhat correct, I'm pretty certain that band-aid is not - 
[image]
[image]
[image]

I there a way to offset the starting construction, or increase "Height" in both directions?

Thanks - 

Bob Travis - Tucson, AZ

looooo, 2022-02-18:
> I there a way to offset the starting construction, or increase "Height" in both directions?

use the module and the height. Not sure if this solves the issue. Maybe I did not understand the question correctly.

Cursor-S, 2026-01-19:
Same issue. Noticed that happens with the `helical`:`beta` as if it's 0 everything goes right, and it's also something with the `base`:`height`.

For testing, I use `module` 2mm, `num_teeth` 60 and `pitch_angle` 56.30993° (90°-arctan(2/3)) and the behaviors below:

beta 20° with height 25mm:
[image]

beta 30° with height 25mm:
[image]

beta 15° with height 25mm (ok):
[image]

beta 15° with height 30mm:
[image]

looooo, 2026-05-31:
> In the meantime I have found that the spiral gearing shape is a loft built of one profile and 19 sections (20 numpoints by default). The loft algorithm can detect related edges from one section to another quite well for low beta values, but with increasing angle it becomes harder to match the right edges. It seems like sometimes it finds corresponding faces of a neighboring tooth and sometimes it creates transitions from top/root faces into flank faces and vice versa.

As far as I can see the sections are positioned properly and so the error seems to be in the lofting algorithm where it selects the edges for each individual face.
In an area from the plane of the defining pitch diameter one tooth width towards the apex (1/3 of the cone distance) the algorithm seems to work without error. We should not increase the module too much to keep the errors outside this area

maybe it would be nice to switch to a sweep instead of a loft to fix this issue. Also I didn't follow any documentation when implementing the helical bevel gears. So I am not sure if beta is used correctly here. Would be nice to have a proper documentation on helical bevel gears (especially spherical involute bevel gears, which is not so often used as it's not so easy to produce)