looooo/freecad.gears #99: Update to new FreeCAD AddonManager metadata schema

luzpaz, 2022-02-18:
https://github.com/FreeCAD/FreeCAD-addons/issues/200

looooo, 2022-02-20:
done: https://github.com/looooo/freecad.gears/commit/6a9baf4fed5ff5982a7ea7c851192a779c925144

luzpaz, 2022-02-20:
Thanks, I'll make a PR to add tags and the Gears bugtracker link. 
Does gears have any external deps?

looooo, 2022-02-20:
> Does gears have any external deps?

yes it depends on pythonocc (optionally), but I would like to change this.

> Also would you consider moving over to using 'master' or 'main' as default branch?

I am planning to merge the develop branch into master and release a 1.0 version. But still there are some tasks open.

luzpaz, 2022-02-20:
Can pythonocc be installed via pip ?

looooo, 2022-02-20:
No, but it's preinstalled in the conda-based bundles.

luzpaz, 2022-02-20:
So gears is broken on non-conda based builds ?

looooo, 2022-02-20:
No, it's only the fillet option which was recently added to the development branch. Also it's optional. No big deal if it is not installed.
Also I wanted to add 2d fillets to the freecad python API. Shouldn't be a big deal, but I am not good with doing such tasks. Would be nice to have some guidance with this task.

luzpaz, 2022-02-20:
> Also I wanted to add 2d fillets to the freecad python API

Is there a way then to bypass using pythonocc as a dependency in that case?
There are more active devs these day, themarkster, 0penBrain, etc... i'm sure if you asked you'd get some feedback

luzpaz, 2023-02-05:
What's the status of this ?