looooo/freecad.gears #108: Gears export to SVG in triplicate

skintigh, 2022-04-14:
A gear made in part designer exports to flattened SVG as one path* **.

A gear made in FCGear WB exports to flattened SVG as 3 gears grouped together. In Inkscape one can ungroup them, drag away one gear that's a single path, then drag away a 2nd that's a single path, then you are left with a 3rd gear that is 4*z disconnected, full-circuit/bi-directional paths (a vector from node A to node B goes A-B-A). This is if you select and export just the gear.

If you export a group of a gear with a axle hole cut out of it and a text label on it, those first 2 single-path gears are gone. All that's left is the doubled-up-path gear made out of 4*z doubled disconnected paths. If you use Inkscape to try join those into 1 path and add a "path effect" like "offset" to compensate for kerf, things get really, really weird. Usually it blinks out of existence, sometimes it becomes a random tangled hairball of looping vectors, sometimes a whole bunch of little balls, sometimes the vectors fly away from each other. 

Grouped or not, if you leave it as is and cut it on a laser cutter, it cuts everything multiple times and in a bizzaro pattern (and axle holes are triple cut, too). I tried cutting some gears like this last night, the laser cutter (double?) cut all the right sides of teeth, then went back and cut all the left sides, and spun around and around the axle holes. The gears seemed to be miscut and warped, possibly from the heat of multiple cuts. 

* Well, I didn't try extruding it first, that might change things.
** With rounded root surfaces/filets, instead of square like in Gear WB. Not sure if that's a good or bad thing, just thought I'd mention the difference.

skintigh, 2022-04-15:
Just tested it with an extruded gear from the part designer. It did the same triplicate nonsense, so maybe this is all a FreeCAD bug.

luzpaz, 2023-02-01:
Can you paste your full freecad [About](https://wiki.freecad.org/About) info and also add an example file (you can zip it so it can be attached to this ticket)