looooo/freecad.gears #109: involuterack incorrectly alters transverse pitch when beta not zero

gcgcgcgc, 2022-05-03:
When creating an involute rack with a non-zero beta it will incorrectly calculate the transverse pitch.

Involute gears with non-zero beta calculate the same transverse pitch as a gear with the same module and zero beta, however involuterack calculates a different transverse pitch and therefore does not mesh with gears of the same configuration.

The transverse pitch sppears to be out by the inverse of cos(beta).
A workaround is to multiply the module of all involuteracks by cos(beta) to correct the incorrect pitch.

looooo, 2022-05-03:
I am not sure if I understand correctly. There is a property to switch between the two modes: `properties_from_tool`. If set to true the base geometry-module is changed when using a beta-angle. Can you have a look if switching this property already solves this issue.

looooo, 2022-05-03:
thanks, yes this makes sense.

closed 2022-05-03 by gcgcgcgc

looooo, 2022-05-03:
https://github.com/looooo/freecad.gears/commit/109df352a9b694be1bfb97ae8925ba3560930fb3