looooo/freecad.gears #113: Wrong Transverse_pitch calculation on involute rack

schwantz69, 2022-09-13:
Hi,
The involute rack have a problem on the transverse_pitch calulation, for example:
Module 3 rack with 10° helix angle calculate 9,428mm but should be 9,5701
Can you fix it ?
Thank you !

looooo, 2022-09-19:
I guess properties_from_tool should have an influence, but I have to think about it :)

schwantz69, 2022-09-21:
I have read another user have open in the past the same topic and he have solved it changing the properties_from_tool in False but for me nothing change.. Is possible that my version is obslolete ? Or the installation of the package are not correct ?
It's a big problem for me.
Thank you for your work

schwantz69, 2022-09-28:
As alternative for solve the problem there is the possibility to re install the previous version of FCgear ? If yes where I can download the zip file ?
Thank you

looooo, 2022-10-04:
There were some issues introduced. I currently don't have the capabilities to fix them.

schwantz69, 2022-10-05:
Thank you for your answer, so please can you send me the instruction for install the older version that was without bug ?

schwantz69, 2022-10-20:
> There were some issues introduced. I currently don't have the capabilities to fix them.

I can understand but please can you tell me where I can found the previous version of Fcgear and how I can install it ?
For me the previous version was ok and I will re install it.

Thank you in advance

looooo, 2022-10-22:
you can use branch `2022` which works for me.

looooo, 2022-10-30:
this should be fixed with https://github.com/looooo/freecad.gears/commit/29e45aa613e0ea0cd6952597f5bbf49dd99270ca

closed 2022-10-30 by looooo

schwantz69, 2022-11-02:
I can confirm that now work correctly !
Many thanks for your great work.