looooo/freecad.gears #120: Calculating root of bevel gear with clearance > 0

pedro-pedrosa, 2023-01-14:
For purposes of cutting a bevel gear I'm trying to figure out the root of the gear at pitch diameter but having trouble doing so as a function of the clearance value. I want to use expressions hence the need for a formula.

I asked this question [on the forum](https://forum.freecadweb.org/viewtopic.php?f=3&t=75238) but I don't know how many people will know about gear math.

After looking at the source code, my understanding is that the teeth points are calculated on a sphere with radius equal to the distance between the gear cone apex and pitch diameter plane. These points are then projected to that plane that is tangent to the sphere.

If this is correct, then I don't quite understand how clearance is taken into account when figuring out the root cut (`r_cut`). The root cut is calculated by a division between a sine and a cosine (which suggests it's trying to project into the tangent plane) but the angles are different since `r_f` takes clearance into account but `z_f` doesn't. Furthermore, if `r_f` is the sine of the dedendum, subtracting the sine of the clearance vector doesn't make sense to me, I believe it should be the cosine here.

I'm currently using this math to make my cuts but I'm posting this here to try to understand where my logic is failing or if there's anything wrong with the addon script.

looooo, 2023-01-19:
I am not sure I understand the question. Is it possible for you to add some drawings explaining what you want to achieve?

pedro-pedrosa, 2023-01-19:
so I have a bevel gear with a 45 degree angle
[image]

I want to cut away some parts of it as shown in this sketch
[image]

[image]

I have 3 construction lines that are concentric at the cone apex representing the addendum angle, pitch angle, and root angle. as you can see these constraints are all expressions that take values like pitch angle etc from the gear parameters. figuring out the addendum angle (49.4924 deg) is quite straightforward, figuring out the dedendum (not shown) is also possible but for the root (leftmost construction line) I also have to take the clearance value into account, but I can't figure out the math for it. right now I'm basically using the same math that is being used in this repo but I don't see how it can be correct.

looooo, 2023-01-19:
I am not sure but I guess I defined the clearence is defined on the base sphere (radius 1) and afterwards it is projected on to the cross sections which are used for the extrusion (loft). I didn't look into this for years, so difficult to say for me if there is something wrong with the math.

pedro-pedrosa, 2023-01-20:
I'd be happy to look into this (measure the actual clearance in a pair of generated meshing gears) and submit a PR, if that's okay?