looooo/freecad.gears #121: Create a helical involute gear on an existing shaft

maxwxyz, 2023-01-15:
Not sure if it is possible (otherwise maybe a feature request):
I am trying to create a helical involute gear on an existing hollow shaft (see: [FreeCAD forum](https://forum.freecadweb.org/viewtopic.php?f=3&t=75255))

Is this possible with the Gears-WB, especially with the "chamfer" or phase out of the teeth into the existing shaft? Is it also possible to create an inner stub shaft spline with Gears-WB on an existing hollow shaft?

looooo, 2023-01-15:
not really. I tried to create something similar with the part-wb and gear-wb, but I guess this is not what you want:
[image]

maxwxyz, 2023-01-15:
not bad, would you share the file or tell me how you did it?

luzpaz, 2024-11-11:
This ticket can be closed.

closed 2024-11-11 by maxwxyz