looooo/freecad.gears #124: Unable to install FCGear in FreeCAD v0.19

amanbop, 2023-03-23:
# In Windows 10, FreeCAD V0.19 :

1. This package is not coming up in Addon Manager External Workbenches list of packages available to install....

2. I tried to install using pip in Windows 10, the folders shown in the attached image got created in site packages.. but the workbench did not show up in the FreeCAD upon start...

3. I removed and installed them to the user/roaming/freecad/mod folder where all the other external workbenches are installed. Again the same folders got created with python code in them but still it did not show up in FreeCAD upon start...

What is the correct way to install it in Windows10, FreeCAD V0.19?

[image]

luzpaz, 2024-09-10:
Is this still relevant ?

luzpaz, 2024-12-22:
@looooo close please

closed 2024-12-22 by looooo