looooo/freecad.gears #127: Crash with older files without newer properties

mhindi2, 2023-04-08:
I have many files (100's) with gears constructed with the FCGears workbench. When I load these with the Gears workbench I get python tracebacks with complaints that traverse_module (and others) not being a property of the gear object. I have fixed all of these by inserting statements like:

``` python
+        if not hasattr(obj, 'traverse_module'):
+            obj.addProperty("App::PropertyLength", "traverse_module", "computed", "traverse module of the generated gear", 1)
+            

```
in features.py before accessing the traverse_module property. Ditto for other properties that were not in the original FCGears workbench gear object. Is it possible to update the code to take into account missing properties for gears built using the FCGears workbench?

jbaehr, 2023-04-26:
Is your issue fixed with PR #126 or does it require more than that?

mhindi2, 2023-04-27:
Yes, it looks like the same issue and it would. I assume _all_ the new properties would be tested for (and there were additional ones besides traverse_module) and fixed as well. It was not 100% clear to me if PR#106 fixed those as well.