looooo/freecad.gears #131: Enveloping worm gear

Gypaets, 2023-08-22:
Feature request: enveloping worm gears. The worm gears currently present in the toolbox are just cylindrical, and it would be cool to have enveloping worm gears as well.

looooo, 2024-01-01:
ref: https://forum.freecad.org/viewtopic.php?t=83797

azalanono, 2024-02-03:
Hello.  
I kindly want to ask if (and when, if not) gears matching a worm gear (the "screw-gear") will be implemented.

Because angle of[ teeth depends on diameter and module of "screw-gear", and also straight teeth do not match a "screw-gear" well, this is a very important but missing thing. almost half-round teeth are not so easy to generate.

I also refer to   https://forum.freecad.org/viewtopic.php?t=83797   as looooo does.

looooo, 2024-02-21:
I tried to solve this, but failed. The problem were  some numeric difficulties with finding the matching "involute" for the different cross sections. Also it's not easy to create the final gear from the computed cross sections. There occt is to blame. At the moment I do not have an idea how to implement the envelop worm gear.

azalanono, 2024-03-04:
Looooo,
I can tell you how I would do it (though it exceeds my programming knowledge, so I can't code that)...

Matching gear sets (the small worm gear and the matching big gear - let's name it wheel) are usually produced by using ththe principle of thread cutters. A "worm gear" shaped milling tool is used to cut the teeth out of the big wheel. This is done in a milling machine. While the cutter is rotating, also the "wheel" is rotated, so the right shape of teeth is the result.
Shouldn't this method be possible in FreeCAD, too?

First, I would generate a short worm gear of 1-2 turns. It should be a little bit oversized to give free play.

Now we have to find the right diameter for the matching wheel. The circumference has to be a multiple of worm gear's pitch. This (pitch multiplied by number of wheel-teeth). This gives the distance (radius) of wheel gear.

Now all you have to do is rotate both (wheel and worm) in synchronous way and remove hereby all excess material from wheel. So the worm gear acts like a cutting tool.

Shouldn't that work in FreeCAD the same way?   Possibly not the fastest method, but can't most of that be already done by FreeCAD? I mean cutting of parts from each other?