looooo/freecad.gears #133: When you make a double helix, the full value does not come

gebze52, 2023-09-11:
When you make a double helix, the surfaces do not give full values, can you fix it?
[image]

looooo, 2023-09-11:
I do not understand. Can you give I sights what's wrong?

gebze52, 2023-09-12:
Sorry, English is not enough for this. I use translation.
Let me explain it this way.
When I make a gear, one face gives full value, the other face is curved and incomplete. It makes an error when mirroring.
The parts I marked in the picture should have a value of 0.
[image]

gebze52, 2023-09-12:
It gives an error at 45 gears. It does not give an error on the first created gear