looooo/freecad.gears #139: No gear_parameters after creating a gear

aerobaticant, 2023-12-19:
I have no gear_parameters appearing in the data tab so I am unable to edit any created gears.
I installed the **freecad.gears workbench** 1.1.0 from the Addon Manager in a clean installation of FreeCAD 0.21.1

```
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 0.21.1.33668 +26 (Git)
Build type: Release
Branch: (HEAD detached at 0.21.1)
Hash: f6708547a9bb3f71a4aaade12109f511a72c207c
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
Locale: English/United Kingdom (en_GB)
Installed mods: 
  * 3D_Printing_Tools
  * fasteners 0.4.75
  * freecad.gears 1.1.0
  * kicadStepUpMod 10.22.4
```

[image]

looooo, 2023-12-24:
I See the parameters on the photo. Not sure what is the issue here

aerobaticant, 2023-12-28:
All of the online examples and videos I have found refer to the **gear_parameters** property which does not exist.
I have now realised that the **base** parameters are what I need to change.
Is there a good reason why the standard **Tasks** panel isn't used when creating / editing gears?

looooo, 2023-12-28:
hmm, no real reason, but there was no need for it (at least for me)

looooo, 2024-01-01:
can we close this one?

aerobaticant, 2024-01-03:
OK I'll close. Can I suggest that anyone who created tutorials update them to match what actually appears.  :-)

closed 2024-01-03 by aerobaticant